clash-lang/clash-compiler #2528: Clash can sometimes create duplicate port names

leonschoorl, 2023-07-03:
```haskell
topEntity :: Bool -> Bool
topEntity x = x
{-# ANN topEntity
  ((defSyn "topEntity") { t_output = PortName "x" }
  ) #-}
```
generates
```vhdl
entity topEntity is
  port(x : in boolean;
       x : out boolean);
end;
```

This is a bit of a weird corner case, as far as I can tell you can only create it by partially specifying port names.

leonschoorl, 2023-07-03:
It seems clash is smart enough to handle the reverse:
```haskell
topEntity :: Bool -> Bool
topEntity result = result
{-# ANN topEntity
  ((defSyn "topEntity") { t_inputs = [PortName "result"] }
  ) #-}
```

```vhdl
entity topEntity is
  port(result   : in boolean;
       result_0 : out boolean);
end;
```

leonschoorl, 2023-07-03:
Instead of trying to be smart about this, maybe we should simple check port names for uniqueness and report an error when they're not unique.

That'll also help users who accidentally use the same port name multiple times.
Because right now clash happily accept:
```haskell
topEntity :: Bool -> Bool
topEntity x = x
{-# ANN topEntity
  ((defSyn "topEntity") { t_inputs = [PortName "foo"], t_output = PortName "foo" }
  ) #-}
```
and generate invalid HDL
```vhdl
entity topEntity is
  port(foo : in boolean;
       foo : out boolean);
end;
```

martijnbastiaan, 2023-07-03:
Duplicate of https://github.com/clash-lang/clash-compiler/issues/2246 ?

leonschoorl, 2023-07-05:
Basically yes, with the addition of https://github.com/clash-lang/clash-compiler/issues/2528#issuecomment-1618561609

closed 2023-07-05 by leonschoorl