clash-lang/clash-compiler #2534: Expose `Clash.Class.Counter` methods

gergoerdi, 2023-07-11:
Currently, functions like `countMin` and `countSuccOverflow` are marked in the documentation as internal implementation details that shouldn't be used by client code. But they (or something similar) would be very useful in a lot of cases where we only want to count "once" and then do something else. 

I realize overflow detection can be achieved "on the outside" by adding one more bit to the counter, without access to these functions by doing something like:

```
let ((bitCoerce @(Index 2) -> overflow), x') = countSucc (0, x)
in x' <$ guard (not overflow)
```

but this is not really readable (it could be improved a bit if we had a `Counter Bool` instance). And I'm not sure what similar trick could apply to `countMin`.

gergoerdi, 2023-07-11:
I think the API I would prefer is one that consists of `countMin`, and `countSuccChecked :: (Counter a) => a -> Maybe a`. Then both `countSucc` and `countSuccOverflow` are trivially recoverable:

```
countSucc = fromMaybe countMin . countSuccChecked
countSuccOverflow = maybe (True, countMin) (False,) . countSuccChecked
```

martijnbastiaan, 2023-07-12:
The nice thing about `(Bool, a)` is that you don't need a branch to determine the overflowed number. While this is a bit of a micro-optimization, I do feel it's the right thing to do.

How about simply moving the internal functions public? I don't think we discovered any downsides to the private API so far, so as far as I'm concerned it's fine as it is.

DigitalBrains1, 2023-07-12:
You could even `bitCoerce` `BitPack a => (Bool, a)` to `Maybe a`, but this comes at quite some performance cost in Haskell simulation (but I note it because it compares to the `bitCoerce` solution already mentioned).

gergoerdi, 2023-10-31:
Are there any plans to implement this?

martijnbastiaan, 2023-11-03:
Would simply exposing the internal function be okay? In that case, it's easily done.

leonschoorl, 2023-11-09:
We could also add an instance for `Counter a => Maybe a`
And then you can get your interface with:
```haskell
countSuccChecked :: Counter a => a -> Maybe a
countSuccChecked = countSucc . Just
```

leonschoorl, 2023-11-09:
Also I we should probably add instances for `Bool` and `Bit`

martijnbastiaan, 2024-03-09:
@gergoerdi I'd be happy to move this forward; could you comment on my replies?

gergoerdi, 2024-03-10:
I'm happy with the suggested ways of doing thidls, because the `Maybe` instance would give me what I need.

martijnbastiaan, 2024-03-10:
See https://github.com/clash-lang/clash-compiler/pull/2692

commit 6659dd84 (2024-03-11): Make functions on `Clash.Class.Counter` public

Fixes #2534
commit 3ed739b7 (2024-03-11): Make functions on `Clash.Class.Counter` public

Fixes #2534
PR #2692 (2024-03-13): Add `Int*`, `Word*`, `Identity` and `Maybe` instances for `Counter`
Fixes #2534

## Still TODO:

  - [X] Write a changelog entry (see changelog/README.md)
  - [X] ~~Check copyright notices are up to date in edited files~~
commit bb74f748 (2024-03-13): Make functions on `Clash.Class.Counter` public

Fixes #2534
closed 2024-03-13 by martijnbastiaan