clash-lang/clash-compiler #2570: Clock blackbox primitive is duplicated

rowanG077, 2023-08-27:
Consider the following `TopEntity`:

```Haskell
{-# LANGUAGE QuasiQuotes #-}

module TopEntity where

import qualified Clash.Prelude as P
import Clash.Explicit.Prelude
import Clash.Annotations.Primitive
import Data.String.Interpolate (i)

oscg :: Clock System
oscg = clockGen
{-# ANN oscg (InlineYamlPrimitive [Verilog] [i|
  BlackBox:
    name: TopEntity.oscg
    kind: Declaration
    template: |-
      // OSCG begin
      OSCG \#( .DIV(6) ) ~GENSYM[oscillator][1] ( .OSC(~RESULT) );
      // OSCG end
  |]) #-}
{-# NOINLINE oscg #-}

topEntity = unsafeFromActiveHigh s1
  where
    clk = oscg

    f = P.exposeClockResetEnable (P.register False) clk noReset enableGen
    s0 = f $ dflipflop clk (pure True)
    s1 = f s0
```

And running `clash TopEntity.hs --verilog` we see that `OSCG` primitive is duplicated 3 times:

```Verilog
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.7.0. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // No inputs

      // Outputs
      output wire  result // reset
    );
  // TopEntity.hs:27:5
  wire  rst;
  reg  result_1 = 1'b0;
  wire  c$app_arg;
  // TopEntity.hs:27:5
  wire  rst_0;
  reg  result_2 = 1'b0;
  wire  c$app_arg_0;
  reg  c$app_arg_1 = ({1 {1'bx}});
  wire  c$app_arg_2;

  assign rst = 1'b0;

  // register begin
  always @(posedge c$app_arg or  posedge  rst) begin : result_1_register
    if ( rst) begin
      result_1 <= 1'b0;
    end else begin
      result_1 <= result_2;
    end
  end
  // register end

  // OSCG begin
  OSCG #( .DIV(6) ) oscillator ( .OSC(c$app_arg) );
  // OSCG end

  assign rst_0 = 1'b0;

  // register begin
  always @(posedge c$app_arg_0 or  posedge  rst_0) begin : result_2_register
    if ( rst_0) begin
      result_2 <= 1'b0;
    end else begin
      result_2 <= c$app_arg_1;
    end
  end
  // register end

  // OSCG begin
  OSCG #( .DIV(6) ) oscillator_0 ( .OSC(c$app_arg_0) );
  // OSCG end

  // delay begin
  always @(posedge c$app_arg_2) begin : c$app_arg_1_delay
    c$app_arg_1 <= 1'b1;
  end
  // delay end

  // OSCG begin
  OSCG #( .DIV(6) ) oscillator_1 ( .OSC(c$app_arg_2) );
  // OSCG end

  assign result = result_1;

endmodule
```
As per @christiaanb suggestion I added a `GHC.Magic.lazy` as a view pattern to the `Clock` argument for:

- https://github.com/clash-lang/clash-compiler/blob/74b68e218e47d774488ddf36b3e8841f00d12fbe/clash-prelude/src/Clash/Signal/Internal.hs#L1411
- https://github.com/clash-lang/clash-compiler/blob/74b68e218e47d774488ddf36b3e8841f00d12fbe/clash-prelude/src/Clash/Signal/Internal.hs#L1454
- https://github.com/clash-lang/clash-compiler/blob/74b68e218e47d774488ddf36b3e8841f00d12fbe/clash-prelude/src/Clash/Signal/Internal.hs#L1474

With no effect.

An interesting observation is that if if rewrite `f` in the failing code from

`f = P.exposeClockResetEnable (P.register False) clk noReset enableGen`

to

`f = register clk noReset enableGen False`

there is no longer any duplication going on.

christiaanb, 2023-08-29:
Does adding a NOINLINE to `clk` work around the problem?

```haskell
topEntity = s1
  where
    clk = oscg
    {-# NOINLINE clk #-}

    f = P.exposeClockResetEnable (P.register False) clk noReset enableGen
    s0 = f $ dflipflop clk (pure True)
    s1 = f s0
```

rowanG077, 2023-08-29:
Well it no longer duplicates the primitive... The primitive is not instantiated at all and clash has made up a clock out of thin air.

```verilog
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.7.0. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // No inputs

      // Outputs
      output wire  result
    );
  // TopEntity.hs:28:5
  wire  rst;
  reg  result_2 = 1'b0;
  reg  c$app_arg = ({1 {1'bx}});
  // TopEntity.hs:28:5
  wire  rst_0;
  reg  result_1 = 1'b0;

  assign rst = 1'b0;

  // register begin
  always @(posedge TopEntity.topEntity_clk or  posedge  rst) begin : result_2_register
    if ( rst) begin
      result_2 <= 1'b0;
    end else begin
      result_2 <= c$app_arg;
    end
  end
  // register end

  // delay begin
  always @(posedge TopEntity.topEntity_clk) begin : c$app_arg_delay
    c$app_arg <= 1'b1;
  end
  // delay end

  assign rst_0 = 1'b0;

  // register begin
  always @(posedge TopEntity.topEntity_clk or  posedge  rst_0) begin : result_1_register
    if ( rst_0) begin
      result_1 <= 1'b0;
    end else begin
      result_1 <= result_2;
    end
  end
  // register end

  assign result = result_1;

endmodule
```

christiaanb, 2023-08-30:
Ah, alright...

So a work-around on GHC 9.0.2 that does work is to eta-expand `f`:

```haskell
topEntity = s1
  where
    clk = oscg

    f x = P.exposeClockResetEnable (P.register False) clk noReset enableGen x
    s0 = f $ dflipflop clk (pure True)
    s1 = f s0
```
produces:
```verilog
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.7.0. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // No inputs

      // Outputs
      output wire  result
    );
  reg  topEntity_f_result = 1'b0;
  reg  topEntity_f_c$app_arg = 1'b0;
  reg  topEntity_f_c$app_arg_0 = ({1 {1'bx}});
  wire  c$app_arg;
  wire  c$app_arg_0;

  // register begin
  always @(posedge c$app_arg_0 or  posedge  c$app_arg) begin : topEntity_f_result_register
    if ( c$app_arg) begin
      topEntity_f_result <= 1'b0;
    end else begin
      topEntity_f_result <= topEntity_f_c$app_arg;
    end
  end
  // register end

  // register begin
  always @(posedge c$app_arg_0 or  posedge  c$app_arg) begin : topEntity_f_c$app_arg_register
    if ( c$app_arg) begin
      topEntity_f_c$app_arg <= 1'b0;
    end else begin
      topEntity_f_c$app_arg <= topEntity_f_c$app_arg_0;
    end
  end
  // register end

  // delay begin
  always @(posedge c$app_arg_0) begin : topEntity_f_c$app_arg_0_delay
    topEntity_f_c$app_arg_0 <= 1'b1;
  end
  // delay end

  assign c$app_arg = 1'b0;

  // OSCG begin
  OSCG #( .DIV(6) ) oscillator ( .OSC(c$app_arg_0) );
  // OSCG end

  assign result = topEntity_f_result;

endmodule
```

leonschoorl, 2023-08-31:
> Well it no longer duplicates the primitive... The primitive is not instantiated at all and clash has made up a clock out of thin air.
That in itself is worrying.
And that seems to happen even on GHC 9.6.2

DigitalBrains1, 2024-11-01:
I think I have two more reproducers that lead to heaps of duplication, with a clock primitive being involved. They generate funky names as well: names like `Duplicate.topEntity_cntr` as well as `\Duplicate.topEntity_cntr`. The latter might technically be valid Verilog, but it seems Vivado isn't too happy about it anyway.

The first variant is without `NOINLINE cntr`:

```haskell
module Duplicate where

import Clash.Explicit.Prelude
import Clash.Explicit.Testbench

topEntity ::
  Signal System (Unsigned 8)
topEntity = cntr + posD
 where
  cntr = register clk1 noReset enableGen 0 0
  posD = register clk2 noReset enableGen 0 0
  done = (== 100) <$> cntr
  (clk1, clk2) = biTbClockGen $ not <$> done
```

This generates the following Verilog:
<details><summary>Duplicate.hs Verilog</summary>

```verilog
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.9.0. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // No inputs

      // Outputs
      output wire [7:0] result
    );
  reg [7:0] c$app_arg = 8'd0;
  // Duplicate.hs:10:3-6
  reg [7:0] \Duplicate.topEntity_cntr  = 8'd0;
  wire  \Duplicate.topEntity_testClk ;
  wire  \Duplicate.topEntity_done ;
  // Duplicate.hs:10:3-6
  reg [7:0] Duplicate.topEntity_cntr = 8'd0;
  wire  Duplicate.topEntity_testClk;
  wire  Duplicate.topEntity_done;
  wire  result;
  wire  c$rst;
  wire  c$rst_0;
  wire  c$rst_1;

  assign c$rst = (1'b0);

  // register begin
  always @(posedge Duplicate.topEntity_testClk or  posedge  c$rst) begin : DuplicatetopEntity_cntr_register
    if ( c$rst) begin
      Duplicate.topEntity_cntr <= 8'd0;
    end else begin
      Duplicate.topEntity_cntr <= 8'd0;
    end
  end
  // register end

  // tbClockGen begin
  // pragma translate_off
  reg  clk;
  // 1 = 0.1ps
  localparam half_period = (100000 / 2);
  always begin
    // Delay of 1 mitigates race conditions (https://github.com/steveicarus/iverilog/issues/160)
    #1 clk =  0 ;
    #1000000 forever begin

      if (~ Duplicate.topEntity_done) begin
        `ifdef VERILATOR
          $c("std::exit(0);");
        `endif
        $finish(0);
      end

      clk = ~ clk;
      #half_period;
      clk = ~ clk;
      #half_period;
    end
  end

  assign Duplicate.topEntity_testClk = clk;
  // pragma translate_on
  // tbClockGen end

  assign Duplicate.topEntity_done = ~ (Duplicate.topEntity_cntr == 8'd100);

  // tbClockGen begin
  // pragma translate_off
  reg  clk_0;
  // 1 = 0.1ps
  localparam half_period_0 = (100000 / 2);
  always begin
    // Delay of 1 mitigates race conditions (https://github.com/steveicarus/iverilog/issues/160)
    #1 clk_0 =  0 ;
    #1000000 forever begin

      if (~ (Duplicate.topEntity_done)) begin
        `ifdef VERILATOR
          $c("std::exit(0);");
        `endif
        $finish(0);
      end

      clk_0 = ~ clk_0;
      #half_period_0;
      clk_0 = ~ clk_0;
      #half_period_0;
    end
  end

  assign result = clk_0;
  // pragma translate_on
  // tbClockGen end

  assign c$rst_0 = (1'b0);

  // register begin
  always @(posedge result or  posedge  c$rst_0) begin : c$app_arg_register
    if ( c$rst_0) begin
      c$app_arg <= 8'd0;
    end else begin
      c$app_arg <= 8'd0;
    end
  end
  // register end

  assign c$rst_1 = (1'b0);

  // register begin
  always @(posedge \Duplicate.topEntity_testClk  or  posedge  c$rst_1) begin : DuplicatetopEntity_cntr_register_0
    if ( c$rst_1) begin
      \Duplicate.topEntity_cntr  <= 8'd0;
    end else begin
      \Duplicate.topEntity_cntr  <= 8'd0;
    end
  end
  // register end

  // tbClockGen begin
  // pragma translate_off
  reg  clk_1;
  // 1 = 0.1ps
  localparam half_period_1 = (100000 / 2);
  always begin
    // Delay of 1 mitigates race conditions (https://github.com/steveicarus/iverilog/issues/160)
    #1 clk_1 =  0 ;
    #1000000 forever begin

      if (~ \Duplicate.topEntity_done ) begin
        `ifdef VERILATOR
          $c("std::exit(0);");
        `endif
        $finish(0);
      end

      clk_1 = ~ clk_1;
      #half_period_1;
      clk_1 = ~ clk_1;
      #half_period_1;
    end
  end

  assign \Duplicate.topEntity_testClk  = clk_1;
  // pragma translate_on
  // tbClockGen end

  assign \Duplicate.topEntity_done  = ~ (\Duplicate.topEntity_cntr  == 8'd100);

  assign result = \Duplicate.topEntity_cntr  + c$app_arg;

endmodule
```
</details>

Al least the following things are duplicated:
```verilog
  reg [7:0] \Duplicate.topEntity_cntr  = 8'd0;
  wire  \Duplicate.topEntity_testClk ;
  wire  \Duplicate.topEntity_done ;
  // Duplicate.hs:10:3-6
  reg [7:0] Duplicate.topEntity_cntr = 8'd0;
  wire  Duplicate.topEntity_testClk;
  wire  Duplicate.topEntity_done;

  assign Duplicate.topEntity_testClk = clk;

  assign Duplicate.topEntity_done = ~ (Duplicate.topEntity_cntr == 8'd100);

  assign result = clk_0;

  assign \Duplicate.topEntity_testClk  = clk_1;

  assign \Duplicate.topEntity_done  = ~ (\Duplicate.topEntity_cntr  == 8'd100);

  assign result = \Duplicate.topEntity_cntr  + c$app_arg;
```
along with 3 instead of 2 clock primitives, and two register blocks for `cntr` instead of 1.

Also really weird: the `done` signal in HDL is `not <$> done` in Haskell?
```verilog
  // tbClockGen begin
  [...]
      if (~ Duplicate.topEntity_done) begin
        `ifdef VERILATOR
          $c("std::exit(0);");
        `endif
        $finish(0);
      end

  assign Duplicate.topEntity_done = ~ (Duplicate.topEntity_cntr == 8'd100);
```

DigitalBrains1, 2024-11-01:
Ah. I did everything I reported about in the previous two messages with GHC 9.8.2. I didn't try other GHC's.