clash-lang/clash-compiler #2593: Bug: HDL generation failure with GHC 9.0.2, multiple hidden, synthesis attribute

DigitalBrains1, 2023-10-21:
In Clash `master`, there is a bug when generating HDL under very specific conditions. Only with GHC 9.0.2 (out of the versions we test in CI), with multiple hidden enabled, an annotation from `Clash.Annotations.SynthesisAttributes` and the following reproducer:

```haskell
topEntity ::
  Signal System Bit ->
  Signal System Bit `Annotate` 'StringAttr "break" "me"
topEntity dIn =
  exposeClockResetEnable (register 0) clk rst enableGen dIn
 where
  (clk, rst) = dummyPll

dummyPll ::
  forall dom .
  KnownDomain dom =>
  (Clock dom, Reset dom)
dummyPll = (clockGen, resetGen)
```

will produce the error:
```
<no location info>: error:
    Clash error call:
    Clash.Netlist.BlackBox.Util(585): IsActiveEnable: Expected Bool or Enable, not: Void Nothing
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/BlackBox/Util.hs:585:13 in clash-lib-1.7.0-inplace:Clash.Netlist.BlackBox.Util
      renderElem, called at src/Clash/Netlist/BlackBox/Util.hs:319:15 in clash-lib-1.7.0-inplace:Clash.Netlist.BlackBox.Util
```

Each of the following is enough to make the code produce HDL again:
* Disable multiple-hidden
* Remove `Annotate`
* Add type signature to either `clk` or `rst` (out of `dummyPll`)
* Eta-reduce `topEntity` (`topEntity = exposeClockResetEnable ...`)
* Make `dummyPll` monomorphic

I created a branch [issue-2593](https://github.com/clash-lang/clash-compiler/tree/issue-2593) which tests the reproducer in CI. The full file with the reproducer is [T2593.hs](https://github.com/clash-lang/clash-compiler/blob/issue-2593/tests/shouldwork/Issues/T2593.hs)

DigitalBrains1, 2023-10-21:
This issue prevents me from writing the code I want to write in the examples in #2592 (I need to add type signatures where they aren't needed).

leonschoorl, 2023-10-24:
When run with `-fclash-debug DebugSilent` or higher you can see it stumble over the type of the enable signal a little earlier:
```
Clash.Normalize(212): Expr belonging to bndr: enableGen (:: Enable
  (ErrOnConflict
     (Signal "System" Bit -> Signal "System" Bit)
     (Merge' (Found "System") (TryDomain (Signal "System" Bit -> Signal "System" Bit) (Signal "System" Bit))))) has a non-representable return type. Not normalising:
(Λdom ->
  Enable @dom (clockGen1[GlobalId] @dom))
  @(ErrOnConflict
      (Signal "System" Bit -> Signal "System" Bit)
      (Merge' (Found "System") (TryDomain (Signal "System" Bit -> Signal "System" Bit) (Signal "System" Bit))))
```

Some alternative workarounds you might like better:
* `exposeClockResetEnable (register 0 dIn) clk rst en`
* `withClockResetEnable clk rst en (register 0 dIn)`

DigitalBrains1, 2023-10-25:
Oh, such a `register 0 dIn` workaround is much smoother, thanks!

closed 2024-02-23 by christiaanb

christiaanb, 2025-01-04:
We've released [v1.8.2](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.2), which includes a fix for this issue.

commit da6ee063 (2024-02-16): Look through annotations in tyfam solving

1. Annotations are implemented as type synonyms.
2. GHC looks through type synonyms when matching type families
3. So Clash should do the same.

Fixes #2593
PR #2673 (2024-02-23): Look through annotations in tyfam solving
1. Annotations are implemented as type synonyms.
2. GHC looks through type synonyms when matching type families
3. So Clash should do the same.

Fixes #2593

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit 1f4c73f7 (2024-02-24): Look through annotations in tyfam solving (#2677)

1. Annotations are implemented as type synonyms.
2. GHC looks through type synonyms when matching type families
3. So Clash should do the same.

Fixes #2593

(cherry picked from commit da6ee06345f24105a9497ebae690a9af0c4ffdf9)