clash-lang/clash-compiler #2634: Separate compilation broken?

bgamari, 2023-12-30:
With `clash-ghc-` and GHC 9.2.8, a simple entity using my [`axi-register` package](https://git.smart-cactus.org/ben/axi-register) builds fine when built in conjunction with `axi-register` in a single compilation:
```
$ clash --vhdl Test.hs -isrc -XRecordWildCards -XUndecidableInstances
GHC: Setting up GHC took: 0.567s
GHC: Compiling and loading modules took: 3.945s
Clash: Parsing and compiling primitives took 0.160s
GHC+Clash: Loading modules cumulatively took 7.201s
Clash: Compiling CDAS.topEntity
Clash: Normalization took 0.002s
Clash: Netlist generation took 0.000s
Clash: Compiling CDAS.topEntity took 0.012s
Clash: Total compilation took 7.214s
```
However, if we use `Cabal` to build `axi-register` and then compile the same testcase using the unfoldings contained in the interface files we have the following error:
```
$ clash --vhdl Test.hs 
GHC: Setting up GHC took: 0.514s
GHC: Compiling and loading modules took: 1.324s
Clash: Parsing and compiling primitives took 0.138s
GHC+Clash: Loading modules cumulatively took 2.071s
Clash: Compiling CDAS.topEntity
Clash: Normalization took 0.010s

<no location info>: error:
    Clash error call:
    Clash.Netlist(1041): Under-applied constructor
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist.hs:1041:17 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
      mkDcApplication, called at src/Clash/Netlist.hs:863:16 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
      mkExpr, called at src/Clash/Netlist.hs:450:30 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
      mkDeclarations', called at src/Clash/Netlist.hs:409:18 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
      mkDeclarations, called at src/Clash/Netlist.hs:284:40 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
      genComponentT, called at src/Clash/Netlist.hs:247:41 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
      genComponent, called at src/Clash/Netlist.hs:125:9 in clash-lib-1.8.0-K7LaVRc69b66OfA18pqV2I:Clash.Netlist
```

This is very reminiscent of https://github.com/clash-lang/clash-compiler/issues/2628

bgamari, 2023-12-30:
I have minimized the issue quite significantly. The data constructor in question here appears to be `Clash.Signal.Internal.(:-)`, which appears to be mentioned a few places in the "bad" case yet not at all in the "good". Given that `:-` isn't synthesizable, is it expected that it appear in programs compiled by Clash?

bgamari, 2023-12-30:
I can confirm that `-fno-worker-worker` avoids the issue.  I have also seen that `clash-cores` disables worker-wrapper. Is this recommended practice? If so, it should probably be documented a bit more prominently.

christiaanb, 2024-01-03:
We do it for the starter project as well, though going by the comment for different reasons: https://github.com/clash-lang/stack-templates/blob/572614b9c34dd4b17f7eb209931e873f220b3a7f/projects/simple/%7B%7Bname%7D%7D.cabal#L56-L59

Also, when we compile from source we emit a warning when W/W is enabled globally: https://github.com/clash-lang/clash-compiler/commit/957a302c9aa3afce17c623e7daa79dfd14dc10fb