clash-lang/clash-compiler #2636: Produced VHDL cannot be used within Vivado block designs

bgamari, 2024-01-11:
It turns out that Vivado's block design facility has the annoyingly arbitrary limitation that ports may only be of type `std_logic` or `std_logic_vector`. Unfortunately, this means that VHDL produced by Clash, which contains various types, cannot be used in this facility.

It would be great if there were a way to flatten ports to `std_logic[_vector]`  to side-step this limitation.

DigitalBrains1, 2024-01-13:
I know we turn a bunch of things into `std_logic_vector` if you specify the compile-time flag `-fclash-hdlsyn Vivado` but I forgot the details. Do you already specify that flag when you observe the problem?

If that flag doesn't do the job, would a flag that turns all ports of the `topEntity` into `std_logic{,_vector}` fit the bill or perhaps on all entities with a `Synthesize` annotation?

bgamari, 2024-01-17:
Indeed I observe the issue with  `-fclash-hdlsyn Vivado`. Sadly `-fclash-no-render-enums` also does not appear to help.

In all cases I see things like:
`clk  : in wombat_types.clk_System;` and `s_axi_awvalid           : in wombat_types.index_2`

martijnbastiaan, 2024-05-30:
A workaround would be to translate to Verilog instead. Whether this is usable entirely depends on your situation of course.