clash-lang/clash-compiler #2656: `unknown RTS option: -xm20000000`

paulyoung, 2024-02-04:
If I try to build `clash-ghc` using Nix, `clash-prelude` fails in the `checkPhase` with `unknown RTS option: -xm20000000` when trying to run `doctests` `unittests`.

I'm using an M1 Macbook Pro.

If I do the following in `nix/overlay.nix` the error goes away.

```diff
       clash-prelude =
-        hprev.callCabal2nixWithOptions
+        hprev.callCabal2nix
           "clash-prelude"
           ../clash-prelude
-          "--flag workaround-ghc-mmap-crash"
           { };
```

I've tried to override this using various Nix mechanisms (e.g. `override`, `overrideAttrs`, a custom overlay, `pkgs.haskell.lib.dontCheck`) and while some of these work when building `clash-prelude` in isolation, it doesn't help when building `clash-ghc` for some reason that I don't understand.

paulyoung, 2024-02-04:
> If I do the following in `nix/overlay.nix` the error goes away.

I spoke too soon. While the build succeeds, actually trying to use `clash` results in a runtime error.

```
nix build .#clash-ghc
./result/bin/clash --help
clash: unknown RTS option: -xm20000000
...
```

paulyoung, 2024-02-04:
I made an additional change to `nix/overlay.nix` and unsurprisingly the runtime error is gone.

```diff
       clash-ghc =
         let
           unmodified =
-            hprev.callCabal2nixWithOptions
+            hprev.callCabal2nix
               "clash-ghc"
-              ../clash-ghc
-              "--flag workaround-ghc-mmap-crash" {
+              ../clash-ghc {
               inherit (hfinal) clash-lib clash-prelude;
             };
```

```
nix build .#clash-ghc
./result/bin/clash --help
Usage:

    clash [command-line-options-and-input-files]

...
```

paulyoung, 2024-02-04:
> I've tried to override this using various Nix mechanisms

To clarify, what I meant here is that I have the flake in this repository as an input to my own flake.

DigitalBrains1, 2024-02-04:
It would seem the `-xm` flag is only on x86_64:
https://gitlab.haskell.org/ghc/ghc/-/blob/0df8ce27f1c418fee1ba860f1c6575f66cae2ca7/rts/RtsFlags.c?page=2#L1756

We had not realised this. We'll need to make it such that the flag is only passed on x86_64.

paulyoung, 2024-02-04:
Thanks! I’ll probably use a fork in the meantime.

I think Rosetta should also allow people to work around this by doing something like:

```nix
…
let
  supportedSystems = [
    flake-utils.lib.system.aarch64-darwin
    flake-utils.lib.system.x86_64-darwin
  ];
in
  flake-utils.lib.eachSystem supportedSystems (
    system: let
      pkgs = import nixpkgs {
        system:
          if system == flake-utils.lib.system.aarch64-darwin
          then flake-utils.lib.system.x86_64-darwin
          else system;
      };
…
```

rowanG077, 2024-02-04:
Should be fixed with https://github.com/clash-lang/clash-compiler/pull/2657 I  run aarch64 but didn't use nix flakes so I didn't hit this before.

commit 22b3a581 (2024-02-04): cabal: Make `workaround-ghc-mmap-crash` a noop on non-x86_64. (#2657)

Fixes #2656
commit 4db6dbf4 (2024-02-04): cabal: Make `workaround-ghc-mmap-crash` a noop on non-x86_64. (#2657) (#2659)

Fixes #2656

(cherry picked from commit 22b3a581a87bf079203302e732008219531ad36b)
commit 0b4733c9 (2024-03-02): cabal: Make `workaround-ghc-mmap-crash` a noop on non-x86_64. (#2657)

Fixes #2656

(cherry picked from commit 22b3a581a87bf079203302e732008219531ad36b)
closed 2024-02-04 by DigitalBrains1