clash-lang/clash-compiler #2675: Add ability to use a different ghc version other than the ghc build clash

evan0greenup, 2024-02-21:
I build ghc via `stack build`. `stack` use its own `ghc` other than system `ghc`. 

As a result, the `ghc` build `clash` will not locate in `PATH` environment variable when `clash` runs.

Is there any way to overcome this issue? (build flag or other option, which configure the build procedures of `clash` without alter my system environment)?

rowanG077, 2024-02-21:
How are you building clash? In general I think the easiest setup is using the stack[-starters](https://github.com/clash-lang/clash-starters) or something close to it. I think if you use that you won't have this issue.

DigitalBrains1, 2024-02-21:
I agree with Rowan, I think you should generally house your Clash design in a starter project. I am also not 100% sure if this is your question.

That said, for quick one-off tests, I do have this:
```
$ cat ~/.local/bin/clash-1.8
#!/bin/sh

unset DIR
DIR="$(pwd)"
cd ~/src/clash/clash-one-eight
stack run --cwd "$DIR" clash -- "$@"
```
```
$ cat ~/.local/bin/clashi-1.8 
#!/bin/sh

unset DIR
DIR="$(pwd)"
cd ~/src/clash/clash-one-eight
stack run --cwd "$DIR" clashi -- "$@"
```
It helps me to run a freestanding `.hs` file with a Clash design. The dir `clash-one-eight` is just an empty starter project, made with
```
$ stack new clash-one-eight clash-lang/simple
$ cd clash-one-eight
$ stack build
```