clash-lang/clash-compiler #2697: Upload `clash-lib`, `clash-ghc`, `clash-prelude-hedgehog` to Stackage (both nightly and lts)

evan0greenup, 2024-03-14:
`clash-prelude` is currently included in Stackage.

It would be nice to add those packages above into Stackage as well to make `stack.yml` without any `extra-deps` fields.

DigitalBrains1, 2024-03-14:
We are painfully aware. Unfortunately, this is impossible: a bug in the version of Cabal prevents building Clash in Stack for GHC's 9.6 and newer. The bug has been fixed, but the fix has not made GHC 9.8.2 in time, so we're hoping it will be included in GHC 9.8.3 in the future so we can re-enter Stackage nightly.

For more information, see for instance https://github.com/clash-lang/clash-compiler/pull/2665#issuecomment-1939044550 .

DigitalBrains1, 2024-03-14:
(Of course the issue is still valid and should remain open even though we can do nothing more towards the resolution. I made an error in judgement when I closed this. Time to grab another coffee.)

christiaanb, 2024-03-14:
That's not the reason we're not included in stackage-nightly, it's because `clash-lib` has a dependency on the officially unmaintained `docopts-hs`.

DigitalBrains1, 2024-03-14:
`docopt` has been in Stackage since [`nightly-2024-03-08`](https://www.stackage.org/nightly-2024-03-08/package/docopt-0.7.0.8). But Martijn and I thought we could not upload Clash because of the Cabal bug. Now when I'm looking at it again, that workaround is only needed for `clash-cores` and `clash-cosim`? So we can just go on Stackage nightly?

I don't know what the process is for getting back into an LTS.

DigitalBrains1, 2024-03-14:
I know the process for getting on Stackage: you make a PR on `master` adding/uncommenting the packages. Sure. That's not the problem. I said _for getting back into an LTS_. My guess is going to be that is not by doing a PR on `master`.

DigitalBrains1, 2024-03-14:
https://github.com/commercialhaskell/stackage/pull/7346 \o/

Woohoo! 🚀

The usual flow for Stackage is that you add/re-add your package to the `nightly`, and when a new LTS is released, that is where your package will be available (again). I do believe I've heard about an alternative flow where you get into the _current_ LTS, but I couldn't find instructions and I don't know what requirements there are for that flow. @evan0greenup, if you _do_ know and also think Clash would qualify, by all means please elucidate me :-D. It would be nice to be in LTS 22.