clash-lang/clash-compiler #2707: `Natural` is not exported from `clash-prelude-1.9.0`'s `Clash.Prelude` GHC 9.0.2 (and lower?)

lmbollen, 2024-04-10:
Found on the `clash-vexriscv` repo:
https://github.com/clash-lang/clash-vexriscv/actions/runs/8615733120/job/23611920320
9.2.8 and 9.4.8 do pass. 
9.0.2:
```
 [4 of 8] Compiling VexRiscv.Random  ( src/VexRiscv/Random.hs, /__w/clash-vexriscv/clash-vexriscv/dist-newstyle/build/x86_64-linux/ghc-9.0.2/clash-vexriscv-0.1/build/VexRiscv/Random.o, /__w/clash-vexriscv/clash-vexriscv/dist-newstyle/build/x86_64-linux/ghc-9.0.2/clash-vexriscv-0.1/build/VexRiscv/Random.dyn_o )

src/VexRiscv/Random.hs:42:16: error:
    Not in scope: type constructor or class ‘Natural’
   |
42 | genNatural :: (Natural, Natural) -> IO Natural
   |                ^^^^^^^

src/VexRiscv/Random.hs:42:25: error:
    Not in scope: type constructor or class ‘Natural’
   |
42 | genNatural :: (Natural, Natural) -> IO Natural
   |                         ^^^^^^^

src/VexRiscv/Random.hs:42:40: error:
    Not in scope: type constructor or class ‘Natural’
   |
42 | genNatural :: (Natural, Natural) -> IO Natural
   |                                        ^^^^^^^
Error: cabal: Failed to build clash-vexriscv-0.1.

Error: Process completed with exit code 1.
````

DigitalBrains1, 2024-04-14:
Ah, we get that as a stray export from `GHC.TypeLits`. I'm not sure it's even supposed to be exported from our Prelude, but we do
https://github.com/clash-lang/clash-compiler/blob/a26bb9ef3cb57f24daec3da30db9fb54af91297a/clash-prelude/src/Clash/Prelude.hs#L139

so we get that. The risks of exporting modules you don't control!

leonschoorl, 2024-05-07:
We're reexporting a lot of things from GHC.TypeLits that we probably shouldn't.
For example since GHC-9.6 this includes: `natSing :: KnownNat n => GHC.TypeNats.SNat n`
Which uses a `SNat` that's different from clash-prelude's own `Clash.Promoted.Nat.SNat`.