clash-lang/clash-compiler #2725: Add `reverseBV` to clash prelude

rowanG077, 2024-05-21:
Slow in simulation implementation:

```Haskell
-- | Reverse bitvector
reverseBV :: KnownNat n => BitVector n -> BitVector n
reverseBV = v2bv . reverse . bv2v
```

1. add this function to `clash-prelude` and
2. make something with better time complexity such as https://graphics.stanford.edu/~seander/bithacks.html#ReverseParallel

_Originally mentioned by @christiaanb in https://github.com/clash-lang/clash-compiler/pull/2694#discussion_r1547859228_

rowanG077, 2024-05-21:
Once in prelude remove `reverseBV` from:

https://github.com/clash-lang/clash-compiler/blob/master/clash-cores/src/Clash/Cores/Crc/Internal.hs#L118-L120

jvnknvlgl, 2024-08-02:
Chiming in here to mention that I'm using the same function here in `Clash.Cores.Sgmii.Common`:

https://github.com/clash-lang/clash-compiler/blob/0b7f4c19642c1bdc1c9ca3d7240789eeb27ba390/clash-cores/src/Clash/Cores/Sgmii/Common.hs#L80-L82

So if something new is added this has to be replaced as well.