clash-lang/clash-compiler #2726: stringsearch is abandoned

locallycompact, 2024-05-27:
clash-lib has a dependency on stringsearch which has been abandoned and has no accessible source on bitbucket.

https://hackage.haskell.org/package/stringsearch

Would the preference here be to fork it from tarball and request maintainership, or replace this library with something else? (Not sure what)

martijnbastiaan, 2024-05-27:
We only use it here:

https://github.com/clash-lang/clash-compiler/blob/e6522d73beb26108aaceb917af7f38e1f978b6ec/clash-lib/tools/v16-upgrade-primitives.hs#L94

There are probably ways to remove that dependency :).

closed 2024-07-26 by martijnbastiaan

christiaanb, 2025-01-04:
We've released [v1.8.2](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.2), which includes a fix for this issue.

commit b589b727 (2024-07-21): Remove `stringsearch` dependency

Fixes #2726
commit 2c786c9a (2024-07-21): Remove `stringsearch` dependency

Fixes #2726
commit 75dbbeb2 (2024-07-26): Remove `stringsearch` dependency (#2765)

Fixes #2726
commit 5c3f1192 (2024-07-26): Remove `stringsearch` dependency (#2765) (#2774)

Fixes #2726

(cherry picked from commit 75dbbeb206b718a39a690a2e198310af9bcd8b6d)