clash-lang/clash-compiler #2736: Enabling -O2 breaks Clash.Magic.clashSimulation

leonschoorl, 2024-06-19:
When running clash on modules compiled by GHC, like in a starter-project setup, enabling `-O2` can break `Clash.Magic.clashSimulation`.
Specifically it's the option `-fspec-constr`, which is implied by `-O2`.

Steps to reproduce:
```
$ stack new my-clash-project clash-lang/simple
```
Change contents of `src/Example/Project.hs` to:
```haskell
{-# OPTIONS_GHC -fspec-constr #-}
-- {-# OPTIONS_GHC -ddump-spec #-}
-- {-# OPTIONS_GHC -ddump-simpl -dverbose-core2core #-}
module Example.Project where

import Clash.Prelude

topEntity :: Int
topEntity
 | clashSimulation = 111
 | otherwise = 222
```

```
$ stack run clash -- Example.Project --vhdl
```
or

```
$ cabal run clash -- Example.Project --vhdl
```

Result:
```vhdl
  result <= to_signed(111,64);
```
We're living in a simulation :scream:

This doesn't happen when running clash directly on the source file.

leonschoorl, 2024-06-19:
```haskell
clashSimulation = True
{-# OPAQUE clashSimulation #-}
```

When enabled `SpecConstr` turns:
```haskell
topEntity = case clashSimulation of
 False -> b
 True -> a
```
into 
```haskell
topEntity = a
```
@christiaanb does this violate the contract of `OPAQUE`?

christiaanb, 2024-06-19:
I guess it follows the letter of the contract, but not the spirit...
```
Note [OPAQUE pragma]
~~~~~~~~~~~~~~~~~~~~
Suppose a function `f` is marked {-# OPAQUE f #-}.  Then every call of `f`
should remain a call of `f` throughout optimisation; it should not be turned
into a call of a name-mangled variant of `f` (e.g by worker/wrapper).
```

Since `clashSimulation` and its content are completely elided in the final result, technically, we see no name-mangled version of `clashSimulation`. i.e. we wouldn't be surprised if something with an `OPAQUE` annotation got elided as part of dead code elimination. That being said, I think in this particular case, GHC is doing something to the `OPAQUE` annoted binder that is non-desirable.

christiaanb, 2024-06-19:
You could try compiling `Clash.Magic` with `{-# OPTIONS_GHC -O0 #-}` to see whether that helps.

martijnbastiaan, 2024-06-20:
Maybe

```haskell
clashSimulation = unsafePerformIO (fromMaybe True <$> lookupEnv "XXX_ISSUE_2736")
{-# OPAQUE clashSimulation #-}
```

christiaanb, 2024-06-20:
Together with @leonschoorl we discoved that
```haskell
import GHC.Magic (noinline)

clashSimulation = noinline True
{-# OPAQUE clashSimulation #-}
```
is a workaround/proper fix. At least it's not something where we need to wait for a new GHC release with a changed behavior with respect to OPAQUE.

rowanG077, 2024-06-20:
Related perhaps: https://github.com/clash-lang/clash-compiler/issues/2570 ?

leonschoorl, 2024-06-21:
@rowanG077 I don't think so, here the problem is happening inside a transformation inside GHC, with #2570 it seems to be inside transformations in clash

PR #2737 (2024-06-30): Make sure clashSimulation is False in HDL even with -O2
Fixes #2736
Also found that the test for #1796 was broken.

## Still TODO:

  - [ ] Write a changelog entry (see changelog/README.md)
  - [ ] Check copyright notices are up to date in edited files

[comment]: # (Depending on the change, some of these points may not be necessary. If so, leave the boxes unchecked so it is easier for a reviewer to see what has been omitted.)
commit 68fb3da2 (2024-06-30): Make sure clashSimulation is False in HDL even with -O2

The SpecConstr pass in GHC is able to poke through the OPAQUE of
clashSimulation.
Adding a noinline seems to block it from seeing the True inside.

Fixes #2736
PR #2746 (2024-06-30): Make sure clashSimulation is False in HDL even with -O2 (copy #2737)
Fixes #2736
Also found that the test for #1796 was broken.

## Still TODO:

  - [ ] Write a changelog entry (see changelog/README.md)
  - [ ] Check copyright notices are up to date in edited files

[comment]: # (Depending on the change, some of these points may not be necessary. If so, leave the boxes unchecked so it is easier for a reviewer to see what has been omitted.)
<hr>This is an automatic copy of pull request #2737 done by [Mergify](https://mergify.com).
commit b305d8d9 (2024-06-30): Make sure clashSimulation is False in HDL even with -O2

The SpecConstr pass in GHC is able to poke through the OPAQUE of
clashSimulation.
Adding a noinline seems to block it from seeing the True inside.

Fixes #2736

(cherry picked from commit 68fb3da233858e8b403e8b2be4db640216b4af8c)
closed 2024-06-30 by martijnbastiaan