clash-lang/clash-compiler #2754: `OverridingBool` prevents auto derivation of various classes

martijnbastiaan, 2024-07-13:
See https://github.com/clash-lang/clash-compiler/pull/2711#discussion_r1670894452. We could remove a whole bunch of code if we were to define our own (isomorphic) version of `OverridingBool`.

leonschoorl, 2024-07-16:
We use the `OverridingBool` because we have to pass it to GHC's api [here](https://github.com/clash-lang/clash-compiler/blob/53aabf2b0d069e85ca794572a91372aaa04bcd41/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L299), possibly in other places too.

Instead of defining our own version and converting from/to GHC's version, we could also just derive the required (orphan) instances for `OverridingBool`.

martijnbastiaan, 2024-07-21:
I don't think that's a good idea. Orphan instances break everyone's code that had the same idea.

commit 03d22d0c (2024-07-21): Add `COverridingBool`

Fixes #2754
commit 4c3bf54f (2024-07-21): Add `COverridingBool`

Fixes #2754
commit 0f390b2b (2024-07-26): Add `COverridingBool`

Fixes #2754
commit 93a54c86 (2024-07-26): Add own version of `OverridingBool`

Fixes #2754