clash-lang/clash-compiler #2757: On the release of `clash-cores`

martijnbastiaan, 2024-07-18:
Tasks:

- [ ] Move `clash-cores/` to `clash-lang/clash-cores`
  - [ ] Create repo
  - [ ] Remove code from `clash-lang/clash-compiler`: note that `clash-testsuite` also refers to `clash-cores`
- [ ] Setup CI on `clash-lang/clash-cores`
- [ ] Decide which cores are experimental / likely to change and mark them as internal/experimental

DigitalBrains1, 2024-07-18:
There are some wrinkles.

The functions in [`Clash.Cores.Xilinx.Interal`](https://github.com/clash-lang/clash-compiler/blob/701867f7025b723804df695179e0fa5188d1ca77/clash-cores/src/Clash/Cores/Xilinx/Internal.hs), to wit `renderTcl`, `IpConfig` and `ReadXdc`, and their friends. These functions can be used to generate the `.clash.tcl` files that are used by the [Clash<->Tcl API](https://hackage.haskell.org/package/clash-lib-1.8.1/docs/Clash-DataFiles.html). These functions should, in my opinion, be moved to `clash-lib`. From a discussion with @martijnbastiaan, I've come to understand we don't actually give any API guarantees for `clash-lib`, so this could actually be partly done here and now with a simple move (after deciding on a module name). But that _would_ mean that `clash-cores` depends on some new version of `clash-lib` that contains these functions.

I bring this up now because as soon as we no longer tightly couple `clash-cores` with `clash-prelude` and `clash-lib`, we might quickly encounter a situation where we need such constructs:
```haskell
#if MIN_VERSION_clash_lib(1,10,0)
import qualified Clash.TclApi as T
#else
import qualified Clash.Cores.Xilinx.Internal as T
#endif
```
and I'd really like to try and avoid that as much as possible.

Another function which also shouldn't be where it currently is, is [`Clash.Cores.Xilinx.Xpm.Cdc.Internal.inst`](https://github.com/clash-lang/clash-compiler/blob/701867f7025b723804df695179e0fa5188d1ca77/clash-cores/src/Clash/Cores/Xilinx/Xpm/Cdc/Internal.hs). And there might be more? We should review what is there in `clash-cores` before deciding to release it.

A different way to avoid these `#if MIN_VERSION` shenanigans is, I think, to state up front that a particular major release X of `clash-cores` only supports a particular major release Y of Clash. That way, you might be able to say:
* `clash-cores` 1.0 only supports Clash 1.8
* `clash-cores` 1.2 only supports Clash 1.10

And that way, `clash-cores` 1.0 would import `Clash.Cores.Xilinx.Internal` above and `clash-cores` 1.2 would import `Clash.TclApi`. No `#if` needed.

(or for clarity, we could set X and Y the same and call the first `clash-cores` release 1.8)

I'm not totally convinced this works. I might have overlooked some intricacy.

DigitalBrains1, 2024-07-18:
A different solution so we can say that `clash-cores` requires GHC 9.4 or up (and we can use the fancy features therein) can be attained by splitting `clash-testsuite` into `clash-testsuite` and `clash-cores-testsuite` (for consistency, continuing the mistake that _testsuite_ is one word). We wouldn't need to split the repositories (although we should do that some day anyway). We could run CI on GHC 8.10 through 9.6 for `clash-testsuite` but only run on GHC 9.4 and 9.6 for `clash-cores-testsuite`.

Note that splitting repositories also requires this split of the test suite...

martijnbastiaan, 2024-07-19:
I don't think `inst` belongs in `clash-lib` at all, and I'm strongly opposed to move it there. I'm not sure if it's written down anywhere, but things in `clash-lib` are like the GHC API: no  guarantees whatsoever. Writing black boxes has been treated as "I hope you know what you're doing" due to all the intracies and weird interactions with GHC. For the Inst stuff I hope to make an API that is stable and is pain-free to use. 

I also don't think the location of `inst` is blocking a release at all. It is clearly marked internal and not present in the Haddock. Over time I'm hoping to move it up the hierarchy, adding the features we need for non-XPM modules. Perhaps we can later split it off into a separate package `clash-inst` or `clash-blackboxes`.

Issues we see with the _public_ API of `clash-cores` should be fixed or moved to an `Experimental` module. Issues with internal APIs should never block a release IMO.

DigitalBrains1, 2024-07-19:
> I don't think `inst` belongs in `clash-lib` at all, and I'm strongly opposed to move it there. 

Which is why I did not suggest it... It sounds like you're disagreeing with me, but I think you might have misinterpreted?

> I also don't think the location of `inst` is blocking a release at all. 

So if you move it to another library, and you support multiple versions of that library, both with and without `inst`, you'll need the `#if MIN_VERSION` stuff, right? That's all I'm saying. I'm saying this is definitely the case for `renderTcl` and might be the case for other stuff as well, such as `inst`.

DigitalBrains1, 2024-07-19:
>  Issues with internal APIs should never block a release IMO.

I'm just not a great fan of all the `LANGUAGE CPP` and `#if MIN_VERSION`. So if we can avoid that, it's better, right? You're going to say, on principle, no, this should never be something to consider, give me all the CPP?

martijnbastiaan, 2024-07-19:
> Which is why I did not suggest it... It sounds like you're disagreeing with me, but I think you might have misinterpreted?

Probably!

I'm not sure I understand your comment concerning inst, CPP, and moving it to a library. If we'd split `inst` of into a library, we'd simply use that library right? 

My intention is to have it internal for a while we test how happy we are with it. Only once we've written _n_ black boxes with it, and are pretty convinced of its API we'd split it off. From that point on we'd have to use CPP (or aggressively up min bounds) on library changes yes. But that's okay right?

DigitalBrains1, 2024-07-19:
I think it would be good if we could merge the fourmolu formatting PR on `clash-compiler` before we split the repo's, so there is no need for another `.git-blame-ignore-revs` file in the new repo. But it's not a really big issue, just a nice-to-have.

DigitalBrains1, 2024-08-01:
Do we want to split off the full history of everything happening in `clash-compiler/clash-cores` into the new repository, using `git subtree`?

Or do we want a new initial commit on the new repo with all the code as it currently is, and direct people to `clash-compiler` for the history before that?

@t-wallet and I both prefer the first.

jvnknvlgl, 2024-08-01:
I have added a `clash-cores` label to mark everything that needs to be moved to the new repository when the time is there.

leonschoorl, 2024-08-28:
There are still many references to clash-cores in the CI infra:

https://github.com/clash-lang/clash-compiler/blob/5927123c1df848190e87cd0830b6a533139ebb39/.github/workflows/ci.yml#L221-L222

https://github.com/clash-lang/clash-compiler/blob/5927123c1df848190e87cd0830b6a533139ebb39/.ci/gitlab/test.yml#L94-L97

https://github.com/clash-lang/clash-compiler/blob/5927123c1df848190e87cd0830b6a533139ebb39/.ci/gitlab/test.yml#L99-L102

https://github.com/clash-lang/clash-compiler/blob/5927123c1df848190e87cd0830b6a533139ebb39/.ci/gitlab/test.yml#L160-L169

That last one is now completely pointless after #2793 got merged.