clash-lang/clash-compiler #2781: Evaluator.instantiate: Not a tylambda: Lambda ...

leonschoorl, 2024-08-05:
```haskell
module BugEvalTyLam
  ( fullMeshSwCcTest
  ) where

import Clash.Explicit.Prelude
import Clash.Cores.Xilinx.Ila (IlaConfig(..), Depth(..), ila, ilaConfig)

fullMeshHwTestDummy ::
  Clock System ->
  (  Signal System Bool
  ,  Vec 1 (Signal System Bool)
  )
fullMeshHwTestDummy sysClk =
  fincFdecIla `hwSeqX`
  ( pure False
  , repeat (pure True)
  )
 where
  fincFdecIla :: Signal System ()
  fincFdecIla = ila
    (ilaConfig ("trigger_0" :> Nil))
    sysClk
    (pure True :: Signal System Bool)

-- | Top entity for this test. See module documentation for more information.
fullMeshSwCcTest ::
  Clock System ->
  (Signal System Bool
  )
fullMeshSwCcTest sysClk = spiDone
 where
  (spiDone
    , ugnsStable -- :: Vec 1 (Signal System Bool) -- this signature causes "Evaluator.instantiate: Not a tylambda: Lambda"
    ) = fullMeshHwTestDummy sysClk
{-# ANN fullMeshSwCcTest (defSyn "fullMeshSwCcTest") #-}
```

Results in:
```
    Evaluator.instantiate: Not a tylambda: Lambda (Id {varName = Name {nameSort = System, nameOcc = "x", nameUniq = 10, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 10, varType = AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.KnownDomain", nameUniq = 8214565720323789993, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (LitTy (SymTy "System")), idScope = LocalId}) (App (Lam (Id {varName = Name {nameSort = System, nameOcc = "x", nameUniq = 9, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 9, varType = AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Cores.Xilinx.Ila.Ila", nameUniq = 8214565720323790114, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (LitTy (SymTy "System"))) (LitTy (SymTy "System")), idScope = LocalId}) (App (Lam (Id {varName = Name {nameSort = System, nameOcc = "x", nameUniq = 8, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 8, varType = AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.Assert", nameUniq = 8214565720323788095, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (AppTy (AppTy (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Data.Type.Ord.OrdCond", nameUniq = 8214565720323788060, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.Bool", nameUniq = 3674937295934324744, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))) (AppTy (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Data.Type.Ord.Compare", nameUniq = 8214565720323788067, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Num.Natural.Natural", nameUniq = 3674937295934324786, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))) (LitTy (NumTy 1))) (LitTy (SymTy "System")))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.True", nameUniq = 3891110078048108586, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.True", nameUniq = 3891110078048108586, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.False", nameUniq = 3891110078048108556, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.TypeError", nameUniq = 3674937295934325098, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Prim.CONSTRAINT", nameUniq = 3674937295934324920, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.BoxedRep", nameUniq = 3891110078048108766, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.Lifted", nameUniq = 3891110078048108808, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.:<>:", nameUniq = 3891110078048108697, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.:<>:", nameUniq = 3891110078048108697, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.:<>:", nameUniq = 3891110078048108697, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.Text", nameUniq = 3891110078048108694, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (LitTy (SymTy "Cannot satisfy: ")))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.ShowType", nameUniq = 3891110078048108703, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Num.Natural.Natural", nameUniq = 3674937295934324786, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))) (LitTy (NumTy 1))))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.Text", nameUniq = 3891110078048108694, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (LitTy (SymTy " <= "))))) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.TypeError.ShowType", nameUniq = 3891110078048108703, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Num.Natural.Natural", nameUniq = 3674937295934324786, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo})))) (LitTy (SymTy "System"))))), idScope = LocalId}) (App (Lam (Id {varName = Name {nameSort = System, nameOcc = "x", nameUniq = 7, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 7, varType = AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Cores.Xilinx.Ila.Internal.IlaConfig", nameUniq = 8214565720323790125, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (LitTy (SymTy "System")), idScope = LocalId}) (App (Lam (Id {varName = Name {nameSort = System, nameOcc = "x", nameUniq = 6, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 6, varType = AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.Clock", nameUniq = 8214565720323789964, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (LitTy (SymTy "System")), idScope = LocalId}) (App (Prim (PrimInfo {primName = "Clash.Cores.Xilinx.Ila.ila#", primType = ForAllTy (TyVar {varName = Name {nameSort = User, nameOcc = "dom", nameUniq = 6989586621679036585, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 6989586621679036585, varType = ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.Symbol", nameUniq = 3674937295934325536, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))}) (ForAllTy (TyVar {varName = Name {nameSort = User, nameOcc = "a", nameUniq = 6989586621679036586, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 6989586621679036586, varType = AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Prim.TYPE", nameUniq = 3674937295934324918, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.BoxedRep", nameUniq = 3891110078048108766, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.Lifted", nameUniq = 3891110078048108808, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))))}) (ForAllTy (TyVar {varName = Name {nameSort = User, nameOcc = "n", nameUniq = 6989586621679036587, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 6989586621679036587, varType = ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Num.Natural.Natural", nameUniq = 3674937295934324786, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))}) (AppTy (AppTy (ConstTy Arrow) (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Signal.Internal.KnownDomain", nameUniq = 8214565720323789993, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))) (VarTy (TyVar {varName = Name {nameSort = User, nameOcc = "dom", nameUniq = 6989586621679036585, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}, varUniq = 6989586621679036585, varType = ConstTy (TyCon (Name {nameSort = User, nameOcc = "GHC.Types.Symbol", nameUniq = 3674937295934325536, nameLoc = UnhelpfulSpan UnhelpfulNoLocationInfo}))})))) (AppTy (AppTy (ConstTy Arrow) (AppTy (AppTy (ConstTy (TyCon (Name {nameSort = User, nameOcc = "Clash.Cores.Xilinx.Ila.Ila", nameUniq = 8214565720323790114, nameLoc = UnhelpfulSpan Unh [... cut ...]
    CallStack (from HasCallStack):
      error, called at src-ghc/Clash/GHC/Evaluator.hs:371:23 in clash-ghc-1.9.0-inplace:Clash.GHC.Evaluator
```

When I initially discovery this bug, it was trigged by the addition of the type signature on `ugnsStable`. Which was weird because it was monomorphic, see [here](https://github.com/bittide/bittide-hardware/commit/fef861041d684d6282f27a118ff457a1308356ab).

But somewhere in minimizing the code it lost that property and now it always produces the error.

christiaanb, 2024-08-05:
Thanks for the small reproducer, I found the culprit quite quickly: https://github.com/clash-lang/clash-compiler/pull/2782

closed 2024-08-27 by christiaanb

christiaanb, 2025-01-04:
We've released [v1.8.2](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.2), which includes a fix for this issue.

commit b535171b (2024-08-26): Fix eta-expansion in evaluator

For some eta-reduced 'e', we used to bogusly eta-expand to:

\x.(\y. e y) x

We now correctly expand to:

\x.\y.(e x) y

Fixes #2781
PR #2782 (2024-08-27): Fix eta-expansion in evaluator
For some eta-reduced 'e', we used to bogusly eta-expand to:
```
\x.(\y. e y) x
```
We now correctly expand to:
```
\x.\y.(e x) y
```
Fixes #2781

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit f9466175 (2024-08-27): Fix eta-expansion in evaluator (#2782)

For some eta-reduced 'e', we used to bogusly eta-expand to:

\x.(\y. e y) x

We now correctly expand to:

\x.\y.(e x) y

Fixes #2781
PR #2800 (2024-08-27): Fix eta-expansion in evaluator (copy #2782)
For some eta-reduced 'e', we used to bogusly eta-expand to:
```
\x.(\y. e y) x
```
We now correctly expand to:
```
\x.\y.(e x) y
```
Fixes #2781

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
<hr>This is an automatic copy of pull request #2782 done by [Mergify](https://mergify.com).