clash-lang/clash-compiler #2787: `Counter` for `Vec`tors

gergoerdi, 2024-08-17:
How about something like this?

```
data Endian = BigEndian | LittleEndian

newtype CounterVec (endian :: Endian) n a = CounterVec{ counters :: Vec n a }
    deriving (Generic, NFDataX, Eq, Show)

instance (Counter a, KnownNat n, 1 <= n) => Counter (CounterVec BigEndian n a) where
    countMin = CounterVec $ repeat countMin
    countMax = CounterVec $ repeat countMax

    countSuccOverflow (CounterVec digits) = (overflow, CounterVec digits')
      where
        (overflow, digits') = mapAccumR f True digits
        f carry d = if carry then countSuccOverflow d else (False, d)

    countPredOverflow (CounterVec digits) = (overflow, CounterVec digits')
      where
        (overflow, digits') = mapAccumR f True digits
        f borrow d = if borrow then countPredOverflow d else (False, d)

instance (Counter a, KnownNat n, 1 <= n) => Counter (CounterVec LittleEndian n a) where
    countMin = CounterVec $ repeat countMin
    countMax = CounterVec $ repeat countMax

    countSuccOverflow (CounterVec digits) = (overflow, CounterVec digits')
      where
        (overflow, digits') = mapAccumL f True digits
        f carry d = if carry then countSuccOverflow d else (False, d)

    countPredOverflow (CounterVec digits) = (overflow, CounterVec digits')
      where
        (overflow, digits') = mapAccumR f True digits
        f borrow d = if borrow then countPredOverflow d else (False, d)
```

Mostly I just want to be able to use a `Counter BigEndian n (Index 10)` as the world's least-effort BCD counter...

martijnbastiaan, 2024-08-19:
Is there a reason we can't build `Counter a => Counter (Vec n a)`? Or do you want to introduce `Counters` as a means to introduce endianness?

gergoerdi, 2024-08-19:
Just neither direction seemed more natural to me than the other, so I went with both.

martijnbastiaan, 2024-08-19:
Yeah, I've had similar thoughts when writing instances for tuples. I see you made https://github.com/clash-lang/clash-compiler/pull/2788 marking this issue as fixed; feel free to open an additional issue proposing to extend `Counter` with endianness if that's important to you.

PR #2788 (2024-09-24): Add `Counter a => Counter (Vec n a)` instance #2787
Fixes: #2787
closed 2024-09-24 by DigitalBrains1