clash-lang/clash-compiler #2789: Clash happily synthesizes literals larger then possible when asked to do so

rowanG077, 2024-08-19:
Clash happily synthesizes the following:

```Haskell
topEntity :: Signal System (Unsigned 3)
topEntity = pure (natToNum @1337)
```

to:

```Verilog
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.8.1. DO NOT MODIFY.
*/
`default_nettype none
`timescale 100fs/100fs
module topEntity
    ( // No inputs

      // Outputs
      output wire [2:0] result
    );

  assign result = 3'd1337;

endmodule
```

While clash does exactly what is asked I think there should be at the very least a warning during synthesis that there is something going wrong here.

christiaanb, 2024-08-19:
I wonder we should create a plugin so that we get warning for simulations as well, similar to what GHC already gives us for e.g. `Int`:
```
ghci> 18446744073709551615 :: Int

<interactive>:4:1: warning: [GHC-97441] [-Woverflowed-literals]
    Literal 18446744073709551615 is out of the Int range -9223372036854775808..9223372036854775807
-1
```

DigitalBrains1, 2024-08-19:
Relevant issue: https://github.com/clash-lang/clash-compiler/issues/2582

rowanG077, 2024-08-19:
Closing this in favor of https://github.com/clash-lang/clash-compiler/issues/2582

closed 2024-08-19 by rowanG077