clash-lang/clash-compiler #2795: More stringy names in Clash.GHC.Evaluator.Primitive

bgamari, 2024-08-22:
Hard-coded stringy names are problematic, especially in GHC 9.10.1 due to the `ghc-internal` rewrite. Some occurrences were handled in https://github.com/clash-lang/clash-compiler/commit/e565d26e4db898c4a6cb0abb98276535a1b0058b, but there are quite a few more in `Clash.GHC.Evaluator.Primitive` (c.f. `GHC.Float.integerToFloat#`).

bgamari, 2024-08-22:
There also appear to be many more such cases in the blackboxes in `clash-lib/prims/common`.

DigitalBrains1, 2024-08-22:
Maybe we can replace the YAML files in `clash-lib/prims` with something like
```haskell
import GHC.Classes (gtInt, geInt, ltInt, leInt)

let
  gtIntName = 'gtInt
  geIntName = 'geInt
  ltIntName = 'ltInt
  leIntName = 'leInt
in InlineYamlPrimitive [minBound..] [__i|
     - BlackBox:
         name: #{gtIntName}
         kind: Expression
         type: 'gtInt :: Int ->
           Int -> Bool'
         template: ~ARG[0] > ~ARG[1]
     - BlackBox:
         name: #{geIntName}
         kind: Expression
         type: 'geInt :: Int ->
           Int -> Bool'
         template: ~ARG[0] >= ~ARG[1]
     - BlackBox:
         name: #{ltIntName}
         kind: Expression
         type: 'ltInt :: Int ->
           Int -> Bool'
         template: ~ARG[0] < ~ARG[1]
     - BlackBox:
         name: #{leIntName}
         kind: Expression
         type: 'leInt :: Int ->
           Int -> Bool'
         template: ~ARG[0] <= ~ARG[1]
     |])
```

somewhere as part of `clash-lib` Haskell files. This a sketch, first of all I don't know if `InlineYamlPrimitive` accepts a list of `BlackBox` and secondly it's still not tied in anywhere anyway, but as a general idea.

Also, we might define an ADT somewhat like
```haskell
data BlackBox =
  { name :: Language.Haskell.TH.Syntax.Name
  , kind :: ...
  , type :: Text
  , template :: Text
  }
```
where we can immediately pass a `Name`. I think this would be a nice way to define primitives in user designs as well, but it might be significantly more effort (not my area of expertise, that part of the code).