clash-lang/clash-compiler #2801: signalAutomaton gets shared, to disastrous results

gergoerdi, 2024-08-27:
The below program runs fine in GHCi, but when compiled and run, fails after the first invocation of `sim`:

```
{-# LANGUAGE ApplicativeDo #-}
module Main where

import Clash.Prelude
import Control.Arrow.Transformer.Automaton
import Data.Maybe

data UpToTwo a
    = None
    | One a
    | Two a a
    deriving (Generic, NFDataX)

circuit :: (HiddenClockResetEnable dom) => Signal dom (Maybe Int) -> Signal dom (Maybe Int)
circuit = moore step extract None
  where
    step s Nothing = None
    step None (Just y) = One y
    step (One x) (Just y) = Two x y
    step (Two x y) (Just z) = One z

    extract (Two x y) = Just $ x + y
    extract _ = Nothing

sim :: Int -> Int -> Int
sim x y = start (signalAutomaton @System circuit) [x, y]
  where
    start (Automaton step) xs = load sim xs
      where
        (_, sim) = step Nothing

    load sim [] = consume sim
    load (Automaton step) (x:xs) = load sim xs
      where
        (_, sim) = step (Just x)

    consume (Automaton step) = fromMaybe (consume sim) output
      where
        (output, sim) = step Nothing

main :: IO ()
main = do
    print $ sim 1 2
    print $ sim 3 4
```

```
3
bug: user error (signalAutomaton: non-affine use of continuation)
```

gergoerdi, 2024-08-27:
On Slack's #Clash, @christiaanb recommended trying `-O0` which indeed is an effective workaround, however, I'm worried what its effect on simulation performance is.

I have also tried adding `{-# OPAQUE sim #-}` as per his recommendation but that does not fix the problem.

rowanG077, 2026-02-18:
I tried various things but I couldn't stop GHC from floating out `signalAutomaton @System circuit` and then sharing it across both sim calls resulting in the error.

I think the real fix is actually making `signalAutomaton` give an `IO (Automaton a b)`

This is a smaller standalone reproducer:

```Haskell
import Data.IORef
import System.IO.Unsafe (unsafePerformIO)

newtype Automaton a b = Automaton { runA :: a -> (b, Automaton a b) }

mk :: () -> Automaton Int Int
mk () = Automaton $ \i -> unsafePerformIO $ do
  let step ref input = do
        old <- atomicModifyIORef' ref (\b -> (True, b))
        if old then fail "non-affine use" else do
          ref' <- newIORef False
          let next = Automaton $ \j -> unsafePerformIO $ step ref' j
          pure (input, next)
  ref0 <- newIORef False
  step ref0 i
{-# OPAQUE mk #-}

sim :: Int -> Int
sim x =
  let (_, s1) = runA (mk ()) 0
      (_, s2) = runA s1 x
      (out, _) = runA s2 0
  in out

main :: IO ()
main = do
  print (sim 1)
  print (sim 3)
```

which fails in the same way. Looking at the core it's clear it floats out `mk ()` which is then re-used:

```Haskell
...
main5 = ((mk ()) `cast` <Co:3> :: ...) main3

main2
  = case main5 of { (ds_d10t, s1_aXM) ->
    case (s1_aXM `cast` <Co:3> :: ...) main4 of { (ds1_d10q, s2_aXY) ->
    case (s2_aXY `cast` <Co:3> :: ...) main3 of
    { (out_aYb, ds2_d10n) ->
    case out_aYb of { I# n_a1f1 -> itos n_a1f1 [] }
    }
    }
    }

main6
  = case main5 of { (ds_d10t, s1_aXM) ->
    case (s1_aXM `cast` <Co:3> :: ...) main7 of { (ds1_d10q, s2_aXY) ->
    case (s2_aXY `cast` <Co:3> :: ...) main3 of
    { (out_aYb, ds2_d10n) ->
    case out_aYb of { I# n_a1f1 -> itos n_a1f1 [] }
    }
    }
    }
...
```