clash-lang/clash-compiler #2809: Verilog and System Verilog code gen bug for `map head`

basile-henry, 2024-09-10:
I first discovered the issue with Clash version 1.4.7 but it seems to still be around with the version of clash documented on `https://clash-lang.org/install/linux/` (stackage lts-19)

Small repro code:

```haskell
import Clash.Prelude

topEntity :: Vec 2 (Vec 2 Bool) -> Vec 2 Bool
topEntity = map head
```

`verilog`:
```
$ stack exec --resolver lts-19 --package clash-ghc -- clash --verilog -outputdir=out Test.hs
Selected snapshot: lts-19.33.
GHC: Setting up GHC took: 0.763s
GHC: Compiling and loading modules took: 0.141s
Clash: Parsing and compiling primitives took 0.224s
GHC+Clash: Loading modules cumulatively took 1.224s
Clash: Compiling Main.topEntity
Clash: Normalization took 0.000s
Clash: Netlist generation took 0.000s

<no location info>: error:
    Clash error call:
    :1:38: error: expected: "[\\", "\\]", "]", "~ACTIVEEDGE", "~AND", "~ARG",
        "~ARGN", "~CMPLE", "~COMPNAME", "~CONST", "~CTXNAME", "~DEPTH", "~DEVNULL",
        "~ENDGENERATE", "~ERROR", "~ERRORO", "~FILE", "~FROMBV", "~GENERATE",
        "~GENSYM", "~INCLUDENAME", "~INDEXTYPE", "~ISACTIVEENABLE", "~ISACTIVEHIGH",
        "~ISINITDEFINED", "~ISLIT", "~ISSCALAR", "~ISSYNC", "~ISUNDEFINED",
        "~ISVAR", "~IW64", "~LENGTH", "~LIT", "~LONGESTPERIOD", "~MAXINDEX",
        "~NAME", "~OTHERSYN", "~OUTPUTWIREREG", "~PERIOD", "~REPEAT", "~RESULT",
        "~SEL", "~SIGD", "~SIGDO", "~SIZE", "~STRCMP", "~SYM", "~TAG", "~TEMPLATE",
        "~TOBV", "~TYP", "~TYPEL", "~TYPM", "~TYPMO", "~TYPO", "~VAR", "~VARS",
        "~VIVADO"
    1 | assign ~RESULT = ~FROMBV[~VAR[vec][0][~SIZE[~TYP[0]]-1 -: ~SIZE[~TYPO]]][~>
      |                                      ^
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/BlackBox/Util.hs:612:5 in clash-lib-1.6.4-80KKrruetCd89p13R34Ejz:Clash.Netlist.BlackBox.Util
```

`system-verilog`:
```
$ stack exec --resolver lts-19 --package clash-ghc -- clash --systemverilog -outputdir=out Test.hs
Selected snapshot: lts-19.33.
GHC: Setting up GHC took: 0.746s
GHC: Compiling and loading modules took: 0.146s
Clash: Parsing and compiling primitives took 0.220s
GHC+Clash: Loading modules cumulatively took 1.225s
Clash: Compiling Main.topEntity
Clash: Normalization took 0.001s
Clash: Netlist generation took 0.000s

<no location info>: error:
    Clash error call:
    :1:38: error: expected: "[\\", "\\]", "]", "~ACTIVEEDGE", "~AND", "~ARG",
        "~ARGN", "~CMPLE", "~COMPNAME", "~CONST", "~CTXNAME", "~DEPTH", "~DEVNULL",
        "~ENDGENERATE", "~ERROR", "~ERRORO", "~FILE", "~FROMBV", "~GENERATE",
        "~GENSYM", "~INCLUDENAME", "~INDEXTYPE", "~ISACTIVEENABLE", "~ISACTIVEHIGH",
        "~ISINITDEFINED", "~ISLIT", "~ISSCALAR", "~ISSYNC", "~ISUNDEFINED",
        "~ISVAR", "~IW64", "~LENGTH", "~LIT", "~LONGESTPERIOD", "~MAXINDEX",
        "~NAME", "~OTHERSYN", "~OUTPUTWIREREG", "~PERIOD", "~REPEAT", "~RESULT",
        "~SEL", "~SIGD", "~SIGDO", "~SIZE", "~STRCMP", "~SYM", "~TAG", "~TEMPLATE",
        "~TOBV", "~TYP", "~TYPEL", "~TYPM", "~TYPMO", "~TYPO", "~VAR", "~VARS",
        "~VIVADO"
    1 | assign ~RESULT = ~FROMBV[~VAR[vec][0][0]][~TYPO];
      |                                      ^
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/BlackBox/Util.hs:612:5 in clash-lib-1.6.4-80KKrruetCd89p13R34Ejz:Clash.Netlist.BlackBox.Util
```

`vhdl` (works):
```
$ stack exec --resolver lts-19 --package clash-ghc -- clash --vhdl -outputdir=out Test.hs
Selected snapshot: lts-19.33.
GHC: Setting up GHC took: 0.683s
GHC: Compiling and loading modules took: 0.134s
Clash: Parsing and compiling primitives took 0.208s
GHC+Clash: Loading modules cumulatively took 1.112s
Clash: Compiling Main.topEntity
Clash: Normalization took 0.001s
Clash: Netlist generation took 0.000s
Clash: Compiling Main.topEntity took 0.018s
Clash: Total compilation took 1.131s
```

I realise that maybe that version is also outdated, but I can still reproduce on `master`:

```
$ nix build .#clash-ghc
basile@flopsy:~/dev/clash-compiler$ CLASH_IGNORE_BROKEN_GHCS=True ./result/bin/clash --verilog -outputdir=out Test.hs
GHC: Setting up GHC took: 0.065s
GHC: Compiling and loading modules took: 0.404s
Clash: Parsing and compiling primitives took 0.235s
GHC+Clash: Loading modules cumulatively took 0.846s
Clash: Compiling Main.topEntity
Clash: Normalization took 0.001s
Clash: Netlist generation took 0.000s

<no location info>: error:
    Clash error call:
    :1:38: error: expected: "[\\", "\\]", "]", "~ACTIVEEDGE", "~AND", "~ARG",
        "~ARGN", "~CMPLE", "~COMPNAME", "~CONST", "~CTXNAME", "~DEPTH", "~DEVNULL",
        "~ENDGENERATE", "~ERROR", "~ERRORO", "~FILE", "~FROMBV", "~GENERATE",
        "~GENSYM", "~INCLUDENAME", "~INDEXTYPE", "~ISACTIVEENABLE", "~ISACTIVEHIGH",
        "~ISINITDEFINED", "~ISLIT", "~ISSCALAR", "~ISSYNC", "~ISUNDEFINED",
        "~ISVAR", "~IW64", "~LENGTH", "~LIT", "~LONGESTPERIOD", "~MAXINDEX",
        "~NAME", "~OTHERSYN", "~OUTPUTUSAGE", "~OUTPUTWIREREG", "~PERIOD",
        "~REPEAT", "~RESULT", "~SEL", "~SIGD", "~SIGDO", "~SIZE", "~STRCMP", "~SYM",
        "~TAG", "~TEMPLATE", "~TOBV", "~TYP", "~TYPEL", "~TYPM", "~TYPMO", "~TYPO",
        "~VAR", "~VARS", "~VIVADO"
    1 | assign ~RESULT = ~FROMBV[~VAR[vec][0][~SIZE[~TYP[0]]-1 -: ~SIZE[~TYPO]]][~>
      |                                      ^
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/BlackBox/Util.hs:666:5 in clash-lib-1.9.0-5CZRWdXreXSLESBdTeAbmI:Clash.Netlist.BlackBox.Util
```

leonschoorl, 2024-09-13:
Thanks for the report.

It's caused by the somewhat weird way that the blackboxes of function arguments to higher order blackboxes are handled.
For some reason, I don't fully understand, they're parsed, then de-parsed back into text, and later re-parsed again.
The [blackbox for head](https://github.com/clash-lang/clash-compiler/blob/5927123c1df848190e87cd0830b6a533139ebb39/clash-lib/prims/verilog/Clash_Sized_Vector.primitives.yaml#L6] contained escaped contains escaped square brackets, which weren't properly escaped when de-parsed, making the later re-parse fail.

closed 2025-09-06 by martijnbastiaan

DigitalBrains1, 2025-11-07:
We've released [v1.8.3](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.3), which includes a fix for this issue.

commit 63b38534 (2024-09-13): Properly de-parse escaped brackets in blackboxes

The way that blackboxes of function arguments to higher order
blackboxes are handled is somewhat weird.
For some reason, I don't fully understand, they're parsed, then
de-parsed back into text, and later re-parsed again.
This de-parsing didn't correctly re-escape previously escaped square
brackets, causing the problem seen in #2809.

Fixes #2809
PR #2989 (2025-09-06): prettyBlackBox: Correctly escape `[` and `]`
The pretty-printer previously failed to round-trip, resulting in invalid primitives due to the logic in `Clash.Netlist.BlackBox.mkFunInput`.

I have also included some clean-ups and error message improvements that I found useful while debugging this.

Fixes #2988.
Fixes #2809.

# TODO
- [x] Add changelog entry
- [x] Add test
commit f2b20fdb (2025-09-06): prettyBlackBox: Correctly escape `[` and `]`

The way that way that blackboxes of function arguments to higher order
blackboxes are handled is somewhat weird.

For historical reasons, they are parsed, pretty-printed, later re-parsed
again. However, the pretty-printer failed to correctly re-escape
previously escaped square brackets, resulting in the generation of
invalid HDL, as seen in #2809.

Fixes #2809.
Fixes #2988.
PR #3001 (2025-09-06): prettyBlackBox: Correctly escape `[` and `]` (copy #2989)
The pretty-printer previously failed to round-trip, resulting in invalid primitives due to the logic in `Clash.Netlist.BlackBox.mkFunInput`.

I have also included some clean-ups and error message improvements that I found useful while debugging this.

Fixes #2988.
Fixes #2809.

# TODO
- [x] Add changelog entry
- [x] Add test<hr>This is an automatic copy of pull request #2989 done by [Mergify](https://mergify.com).