clash-lang/clash-compiler #2827: SaturatingNum Bit

gergoerdi, 2024-10-10:
I have a use case where `Overflowing Bit`, and thus a `SaturatingNum Bit` instance, would be useful. I see no reason `Bit` couldn't be made into `SaturatingNum`. Sketch(y) implementation:

```
instance SaturatingNum Bit where
    satAdd = \case
        SatWrap -> addOr 0
        SatBound -> addOr maxBound
        SatZero ->  addOr 0
        SatSymmetric -> addOr maxBound
        SatError -> \_ _ -> errorX "satAdd"
      where
        addOr ext = \x y -> if x == 1 && y == 1 then ext else x + y

    satSub = \case
        SatWrap -> subOr 1
        SatBound -> subOr minBound
        SatZero ->  subOr 0
        SatSymmetric -> subOr maxBound
        SatError -> \_ _ -> errorX "satSub"
      where
        subOr ext = \x y -> if x == 0 && y == 1 then ext else x - y

    satMul = \case
        _ -> (*)
```

kleinreact, 2024-10-11:
As also mentioned on [#clash](https://functionalprogramming.slack.com/archives/CPGMJFF50), another alternative might be to use the already existing instance for `BitVector 1`, which probably results in more maintainable code, e.g.,

```haskell
instance SaturatingNum Bit where
  satAdd mode a b = bitCoerce @(BitVector 1) @Bit $
    satAdd mode
      (bitCoerce @Bit @(BitVector 1) a)
      (bitCoerce @Bit @(BitVector 1) b)

  satSub mode a b = bitCoerce @(BitVector 1) @Bit $
    satSub mode
      (bitCoerce @Bit @(BitVector 1) a)
      (bitCoerce @Bit @(BitVector 1) b)

  satMul mode a b = bitCoerce @(BitVector 1) @Bit $
    satMul mode
      (bitCoerce @Bit @(BitVector 1) a)
      (bitCoerce @Bit @(BitVector 1) b)
```

gergoerdi, 2024-10-11:
```
  satAdd mode `on` bitCoerce @Bit @(BitVector 1)
```

should work with `Data.Function.on`, for that sweet sweet code golfing:)

DigitalBrains1, 2024-10-11:
I'm not a big fan of using the `pack` function here; it is expensive in simulation. It also turns an `XExcepton` into an undefined bit, but I have no problem with that; in my opinion, a `Bit` shouldn't carry an `XException` value and if it does, I have no problem with turning it into what I think is the correct encoding: an undefined bit.

I think the code from the first post is fine (although I'd write `satMul` differently myself).

DigitalBrains1, 2024-10-11:
Ah, but the code in the first post turns an undefined bit into an `XException` X-D

So yeah, I'm in favour of code that doesn't use `pack` and also doesn't produce `XException`s. That'd be my ideal, but maybe the route through `pack` is forgivable. I'd rather not, though.

DigitalBrains1, 2024-10-12:
The implementation for `satMul` is nicely symmetric with `satAdd` and `satSub`, but I still think `satMul _ = (*)` makes more sense. I'm not sure if the mode argument should be forced to WHNF.

Also, multiplication for `Bit` is actually smarter than all of `(*)`, `mul` and `satMul` for `BitVector`. The `BitVector` ones don't notice when the least significant bits are known to be zero despite the presence of undefined bits (there's even more that could be determined, but that one is easy).

I think we should also discuss (elsewhere) why those `BitVector` functions throw `XException` instead of setting undefined bits.

```haskell
>>> xbv = $(bLit ".")
>>> xb = unpack xbv :: Bit
>>> xbv
0b.
>>> xb
.
>>> 1 * xb
.
>>> 0 * xb
0
>>> xb * 0
0
>>> 0 * xbv
*** Exception: X: Clash.Sized.BitVector.* called with (partially) undefined arguments: 0b0 * 0b.
```