clash-lang/clash-compiler #2834: Verilog generation runs indefinitely

kleinreact, 2024-10-28:
Clash seems to diverge on the following reproducer (if being called with the `--verilog` flag).

```haskell
-- diverges for: Size > 1
type Size = 2

topEntity :: Bit
topEntity = head table

table :: Vec Size Bit
table = lookupTable
 where
  lookupTable :: Vec Size Bit
  lookupTable = smap fill $ repeat ()

  fill :: forall n. SNat n -> () -> Bit
  fill n@(SNat :: SNat n) _ = case compareSNat n (SNat @0) of
    SNatLE -> low
    SNatGT -> case compareSNat n (SNat @Size) of
      SNatLE -> at @(n - 1) @(Size - n) SNat lookupTable
      SNatGT -> error "impossible"
```

kleinreact, 2024-10-28:
The compilation gets stuck in an infinite loop when calling [normalizeTopLvlBndr](https://github.com/clash-lang/clash-compiler/blob/master/clash-lib/src/Clash/Normalize/Util.hs#L405) on `table[8214565720323799906]`, which is bound to the term referenced below.

<details><summary>Click here to see the term that is bound to <code>table[8214565720323799906]</code>.</summary>

```
Clash.Sized.Vector.reverse @2
  @Clash.Sized.Internal.BitVector.Bit[8214565720323789991]
  (Clash.Sized.Vector.dfold
     @(Clash.Sized.Vector.VCons[8214565720323790001]
         Clash.Sized.Internal.BitVector.Bit[8214565720323789991])
     @2
     @GHC.Tuple.Unit[3746994889972252672]
     Test.table1[8214565720323804889][GlobalId]
     (GHC.Internal.Data.Proxy.Proxy[8214565720323796227]
        @(Data.Singletons.TyFun[8214565720323800214]
            GHC.Num.Natural.Natural[3674937295934324786]
            (GHC.Prim.TYPE[3674937295934324918]
               (GHC.Types.BoxedRep[3891110078048108766]
                  GHC.Types.Lifted[3891110078048108808]))
        -> GHC.Prim.TYPE[3674937295934324918]
             (GHC.Types.BoxedRep[3891110078048108766]
                GHC.Types.Lifted[3891110078048108808]))
        @(Clash.Sized.Vector.VCons[8214565720323790001]
            Clash.Sized.Internal.BitVector.Bit[8214565720323789991]))
     (Λl[6989586621679028560] ->
     λ(sn[6989586621679028561] :: Clash.Promoted.Nat.SNat[8214565720323789861]
                                    l[6989586621679028560]) ->
     λ(x[6989586621679028562] :: GHC.Tuple.Unit[3746994889972252672]) ->
     λ(xs'[6989586621679028563] :: Data.Singletons.Apply[8214565720323800213]
                                     GHC.Num.Natural.Natural[3674937295934324786]
                                     (GHC.Prim.TYPE[3674937295934324918]
                                        (GHC.Types.BoxedRep[3891110078048108766]
                                           GHC.Types.Lifted[3891110078048108808]))
                                     (Clash.Sized.Vector.VCons[8214565720323790001]
                                        Clash.Sized.Internal.BitVector.Bit[8214565720323789991])
                                     l[6989586621679028560]) ->
     Clash.Sized.Vector.Cons[8214565720323790005]
       @(GHC.Internal.TypeNats.+[3674937295934325540]
           l[6989586621679028560]
           1)
       @Clash.Sized.Internal.BitVector.Bit[8214565720323789991]
       @l[6989586621679028560]
       (_CO_
          @(GHC.Prim.~#[3674937295934324842]
              GHC.Num.Natural.Natural[3674937295934324786]
              GHC.Num.Natural.Natural[3674937295934324786]
              (GHC.Internal.TypeNats.+[3674937295934325540]
                 l[6989586621679028560]
                 1)
              (GHC.Internal.TypeNats.+[3674937295934325540]
                 l[6989586621679028560]
                 1)))
       (let
          wild[6341068275337658369] :: Clash.Promoted.Nat.SNat[8214565720323789861]
                                         l[6989586621679028560]
          = sn[6989586621679028561][LocalId]
       in case wild[6341068275337658369][LocalId] of
            Clash.Promoted.Nat.SNat[8214565720323789862]
              ($dKnownNat[6989586621679026853] :: GHC.Internal.TypeNats.KnownNat[3602879701896396842]
                                                    l[6989586621679028560]) ->
              case Clash.Promoted.Nat.compareSNat[8214565720323788356][GlobalId]
                     @l[6989586621679028560]
                     @0
                     wild[6341068275337658369][LocalId]
                     (Clash.Promoted.Nat.SNat[8214565720323789862] @0
                        (GHC.Num.Natural.NS 0##)) of
                Clash.Promoted.Nat.SNatLE[8214565720323789865]
                  (irred[6989586621679026872] :: GHC.Internal.TypeError.Assert[8214565720323788133]
                                                   (GHC.Internal.Data.Type.Ord.OrdCond[8214565720323788087]
                                                      GHC.Types.Bool[3674937295934324744]
[... 177 lines cut ...]
                              (GHC.Types.I# 22#))
                           GHC.Internal.Stack.Types.EmptyCallStack[8214565720323801273])
                        (GHC.CString.unpackCString# "impossible"))
       xs'[6989586621679028563][LocalId])
     ((Λb[6989586621679026138] ->
        Clash.Sized.Vector.Nil[8214565720323790006] @0
          @b[6989586621679026138]
          (_CO_
             @(GHC.Prim.~#[3674937295934324842]
                 GHC.Num.Natural.Natural[3674937295934324786]
                 GHC.Num.Natural.Natural[3674937295934324786]
                 0
                 0)))
        @Clash.Sized.Internal.BitVector.Bit[8214565720323789991])
     (Clash.Sized.Vector.reverse @2
        @GHC.Tuple.Unit[3746994889972252672]
        (Clash.Sized.Vector.replicate @2
           @GHC.Tuple.Unit[3746994889972252672]
           (Clash.Promoted.Nat.SNat[8214565720323789862] @2
              Test.table1[8214565720323804889][GlobalId])
           GHC.Tuple.()[3963167672086036480])))
```
</details>

My guess is that the problem is introduced by the recursive call to `table[8214565720323799906]` in the `splitAt` call, which results from the `at` call in the example above.

kleinreact, 2024-10-28:
Moving the local definitions to the top level also seems to solve the issue, as the following terminates as expected
```haskell
type Size = 2

topEntity :: Bit
topEntity = head lookupTable

lookupTable :: Vec Size Bit
lookupTable = smap fill $ repeat ()

fill :: forall n. SNat n -> () -> Bit
fill n@(SNat :: SNat n) _ = case compareSNat n (SNat @0) of
  SNatLE -> low
  SNatGT -> case compareSNat n (SNat @Size) of
    SNatLE -> at @(n - 1) @(Size - n) SNat lookupTable
    SNatGT -> error "impossible"
```

kleinreact, 2024-10-28:
Adding a `{-# NOINLINE fill #-}` also makes it work.

christiaanb, 2024-10-29:
The problem again seems to start in the same place as https://github.com/clash-lang/clash-compiler/issues/2831#issuecomment-2444254765

Where  `caseCon` will not reduce the case-expression of the form:
```
case compareSNat d0 d0 of ...
```
because the subject of the case expression is not a data constructor and a primitive.

As a result, the Clash compiler will evaluate both the subject and the alternatives in "parallel". This then causes the `at @1 @1 d1 lookuptable` to be expanced forever, as the branch is never pruned before the "parallel" evaluation of the subject and the alternatives is again performed.