clash-lang/clash-compiler #2851: `enumFrom` errors for `Index 0`

kleinreact, 2024-11-27:
`enumFrom` and alike should not error for `Index 0`. They should return the empty list instead.
```
clashi> enumFrom (undefined :: Index 0)
*** Exception: X: maxBound of 'Index 0' is undefined
```

christiaanb, 2024-11-27:
I think the definition of `Enum` isn't clear that  it _should not_ error in  this case, or that it _should_ return the empty list. I agree it would be nice and that we should probably change the implementation to behave accordingly.

christiaanb, 2024-11-27:
I mean, it even states https://hackage.haskell.org/package/base-4.20.0.1/docs/Data-Enum.html#v:enumFrom

> a possible implementation being enumFrom n = n : enumFrom (succ n)

which would have the following reduction to WHNF:

```
enumFrom (undefined :: Index 0)
==>
(undefined :: Index 0) : enumFrom (succ (undefined :: Index 0))
```

DigitalBrains1, 2024-11-27:
I agree that the empty list is the desirable outcome here, even though there are no Haskell laws preventing the current implementation.