clash-lang/clash-compiler #2852: clash-lib-1.8.1 failed to build with ghc-9.8.4 Stackage Nightly

juhp, 2024-12-05:
```
       /home/curators/work/unpack-dir/unpacked/clash-lib-1.8.1-03ec2cb411fe7ac811914edffea21995ffd0d5c5b6898264fc548c766d146ecb/src/
Clash/Core/Type.hs:516:20: error: [GHC-83865]                                                                                       
           • Couldn't match type ‘GHC.Word.Word64’ with ‘Int’                                                                       
             Expected: Clash.Unique.Unique                                                                                          
               Actual: GHC.Word.Word64                                                                                              
           • In the second argument of ‘(==)’, namely                                                                               
               ‘getKey typeNatAddTyFamNameKey’                                                                                      
             In the expression: nameUniq tc == getKey typeNatAddTyFamNameKey                                                        
             In a stmt of a pattern guard for                                                                                       
                            an equation for ‘reduceTypeFamily’:                                                                     
               nameUniq tc == getKey typeNatAddTyFamNameKey                                                                         
           |                                                                                                                        
       516 |   | nameUniq tc == getKey typeNatAddTyFamNameKey                                                                       
           |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                       
                                                                                                                                    
       /home/curators/work/unpack-dir/unpacked/clash-lib-1.8.1-03ec2cb411fe7ac811914edffea21995ffd0d5c5b6898264fc548c766d146ecb/src/
Clash/Core/Type.hs:521:20: error: [GHC-83865]                                                                                       
           • Couldn't match type ‘GHC.Word.Word64’ with ‘Int’                                                                       
             Expected: Clash.Unique.Unique                                                                                          
               Actual: GHC.Word.Word64                                                                                              
           • In the second argument of ‘(==)’, namely
               ‘getKey typeNatMulTyFamNameKey’
             In the expression: nameUniq tc == getKey typeNatMulTyFamNameKey
             In a stmt of a pattern guard for
                            an equation for ‘reduceTypeFamily’:
               nameUniq tc == getKey typeNatMulTyFamNameKey
           |
       521 |   | nameUniq tc == getKey typeNatMulTyFamNameKey
           |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        
       /home/curators/work/unpack-dir/unpacked/clash-lib-1.8.1-03ec2cb411fe7ac811914edffea21995ffd0d5c5b6898264fc548c766d146ecb/src/
Clash/Core/Type.hs:526:20: error: [GHC-83865]
           • Couldn't match type ‘GHC.Word.Word64’ with ‘Int’
             Expected: Clash.Unique.Unique
               Actual: GHC.Word.Word64
           • In the second argument of ‘(==)’, namely
               ‘getKey typeNatExpTyFamNameKey’
             In the expression: nameUniq tc == getKey typeNatExpTyFamNameKey
             In a stmt of a pattern guard for
                            an equation for ‘reduceTypeFamily’:
                            an equation for ‘reduceTypeFamily’:                                                01:50:14 [313/937727]
               nameUniq tc == getKey typeNatExpTyFamNameKey                                                                         
           |
       526 |   | nameUniq tc == getKey typeNatExpTyFamNameKey
           |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        
       /home/curators/work/unpack-dir/unpacked/clash-lib-1.8.1-03ec2cb411fe7ac811914edffea21995ffd0d5c5b6898264fc548c766d146ecb/src/
Clash/Core/Type.hs:531:20: error: [GHC-83865]
           • Couldn't match type ‘GHC.Word.Word64’ with ‘Int’
             Expected: Clash.Unique.Unique
               Actual: GHC.Word.Word64
           • In the second argument of ‘(==)’, namely
               ‘getKey typeNatSubTyFamNameKey’
             In the expression: nameUniq tc == getKey typeNatSubTyFamNameKey
             In a stmt of a pattern guard for
                            an equation for ‘reduceTypeFamily’:
               nameUniq tc == getKey typeNatSubTyFamNameKey
           |
       531 |   | nameUniq tc == getKey typeNatSubTyFamNameKey
           |                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        
       /home/curators/work/unpack-dir/unpacked/clash-lib-1.8.1-03ec2cb411fe7ac811914edffea21995ffd0d5c5b6898264fc548c766d146ecb/src/
Clash/Core/Type.hs:552:20: error: [GHC-83865]
           • Couldn't match type ‘GHC.Word.Word64’ with ‘Int’
             Expected: Clash.Unique.Unique
               Actual: GHC.Word.Word64
           • In the second argument of ‘(==)’, namely
               ‘getKey typeNatCmpTyFamNameKey’
             In the expression: nameUniq tc == getKey typeNatCmpTyFamNameKey
[... 152 lines cut ...]
               Actual: GHC.Word.Word64
           • In the third argument of ‘Name’, namely
               ‘(getKey ordEQDataConKey)’
             In the expression:
               Name User (showt 'EQ) (getKey ordEQDataConKey) wiredInSrcSpan
             In a case alternative:
                 EQ -> Name User (showt 'EQ) (getKey ordEQDataConKey) wiredInSrcSpan
           |
       579 |             EQ -> Name User (showt 'EQ) (getKey ordEQDataConKey) wiredInSrcSpan
           |                                          ^^^^^^^^^^^^^^^^^^^^^^
        
       /home/curators/work/unpack-dir/unpacked/clash-lib-1.8.1-03ec2cb411fe7ac811914edffea21995ffd0d5c5b6898264fc548c766d146ecb/src/
Clash/Core/Type.hs:580:42: error: [GHC-83865]
           • Couldn't match type ‘GHC.Word.Word64’ with ‘Int’
             Expected: Clash.Unique.Unique
               Actual: GHC.Word.Word64
           • In the third argument of ‘Name’, namely
               ‘(getKey ordGTDataConKey)’
             In the expression:
               Name User (showt 'GT) (getKey ordGTDataConKey) wiredInSrcSpan
:
:
```

Not sure what changed

leonschoorl, 2024-12-23:
The widening of the GHC uniques got [backported to GHC-9.8.4](https://gitlab.haskell.org/ghc/ghc/-/merge_requests/13624).

We already handle this [for GHC-9.10](https://github.com/clash-lang/clash-compiler/commit/1f8e7658b0c50ac682b425b4b4d7b8ef8c80d722#diff-4bcf3059871a130741e98422e240775122d1b38a4edd3952c92b5f977f296d58R37-R43), but [not in a way that's safe on 32bit systems](https://github.com/clash-lang/clash-compiler/blob/8a827e1e232b795a92da584c3e9cd51a46315680/changelog/2024-07-26T07_36_58%2B02_00_ghc_910_upgrade)

closed 2025-01-02 by christiaanb

christiaanb, 2025-01-04:
We've released [v1.8.2](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.2), which includes a fix for this issue.

PR #2860 (2025-01-02): Add support for GHC 9.8.4
Fixes #2852

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
commit d2097b36 (2025-01-02): Add support for GHC 9.8.4

Fixes #2852
commit 4212d036 (2025-01-02): Add support for GHC 9.8.4

Fixes #2852

(cherry picked from commit d2097b36d481bcccf3f2fd6ac5b44072461cfcc6)