clash-lang/clash-compiler #2861: The `Ord` instance of `BitVector` lacks undefined bits support

kleinreact, 2025-01-02:
Using comparison on `BitVector`s  with undefined bits always causes simulation to crash, although the result is well defined in the presence of undefined bits.

To give an example, let's say that I have a custom data type:
```haskell
data T = B Bool | D (Unsigned 2) deriving (Generic, BitPack, Eq, Ord)
```

Then using the derived comparison causes quite some extra code to be generated, e.g.,
```haskell
topEntity :: T -> T -> Bool
topEntity a b = a <= b
```
produces 

<details>
<summary>Click to expand</summary>

```verilog
module topEntity
    ( // Inputs
      input wire [2:0] a
    , input wire [2:0] b
      // Outputs
    , output wire  result
    );
  wire  c$case_alt;
  wire  c$case_alt_0;
  wire  c$case_alt_1;
  wire  c$case_alt_2;
  wire  c$case_alt_3;
  wire  a1;
  wire  b1;
  wire [1:0] a1_0;
  wire [1:0] b1_0;

  assign c$case_alt = b1 ? 1'b0 : 1'b1;
  assign c$case_alt_0 = a1 ? 1'b1 : c$case_alt;
  assign c$case_alt_1 = (a1_0 < b1_0) ? 1'b0 : 1'b1;
  assign c$case_alt_2 = a[2:2] ? 1'b0 : c$case_alt_0;
  assign c$case_alt_3 = a[2:2] ? c$case_alt_1 : 1'b1;
  assign a1 = b[1:1];
  assign b1 = a[1:1];
  assign a1_0 = b[1:0];
  assign b1_0 = a[1:0];
  assign result = b[2:2] ? c$case_alt_3 : c$case_alt_2;
endmodule
```
</details>

Thus, it might be preferable to directly compare the packed representation instead, e.g.,
```haskell
topEntity :: BitVector (BitSize T) -> BitVector (BitSize T) -> Bool
topEntity a b = pack a <= pack b
```
which just produces
```verilog
module topEntity
    ( // Inputs
      input wire [2:0] a
    , input wire [2:0] b
      // Outputs
    , output wire  result
    );
  assign result = (a) <= (b);
endmodule
```
However, this latter `topEntity` will error in simulation, e.g.,
```
clashi> (B True) <= (D 0)
True
clashi> pack (B True) <= pack (D 0)
*** Exception: X: Clash.Sized.BitVector.<= called with (partially) undefined arguments: 0b01. <= 0b100
```

Note that although `0b01.` contains an undefined bit here, it does not matter whether that bit evaluates to `0` or `1`, the result will always be `True`. Hence, comparison will behave well in hardware, while it does not in Clash simulation.

DigitalBrains1, 2025-01-02:
There's lots of stuff we could do more intelligently with `BitVector` on undefined bits. For instance `.. * .0 = .0`, `01. + 010 = 10.`, `00. + 00. = 0..` etcetera. I think those are even more important, because currently we have that `(&& 0)` is smarter than `(* 0)`, and even that `Bit` is smarter than `BitVector 1`, which I all find somewhat grating. Before we implement them, we should watch out that we're not being smarter than any HDL synthesis tools (or we would no longer produce the same defined results, which is a requirement for us).

By the way, I think you just omitted this because it is tangential, but I wouldn't do `pack a <= pack b`, I'd just define a custom `Ord T` that did that after I fully verified the two are equivalent.

I do feel you're cheating here, I wonder if you even noticed. Because you changed the type of the top entity, so you just got rid of `T` altogether. The `pack` in there is actually `BitVector -> BitVector`. If I keep the type of the top entity as containing `T`, I get this (Clash 1.8.1, GHC 9.4.7):

<details><summary><code>pack a <= pack b</code></summary>

```verilog
module topEntity
    ( // Inputs
      input wire [2:0] a
    , input wire [2:0] b

      // Outputs
    , output wire  result
    );
  wire [2:0] c$app_arg;
  // OrdBV.hs:5:53-59
  wire [1:0] g1;
  // OrdBV.hs:5:53-59
  wire  g1_0;
  wire [0:0] c$app_arg_0;
  wire [2:0] result_1;
  wire [2:0] c$app_arg_1;
  // OrdBV.hs:5:53-59
  wire [1:0] g1_1;
  // OrdBV.hs:5:53-59
  wire  g1_2;
  wire [0:0] c$app_arg_2;
  wire [2:0] result_2;

  assign c$app_arg = b[2:2] ? ({1'b1,(((g1)))}) : ({1'b0,({(c$app_arg_0),1'bx})});

  assign g1 = b[1:0];

  assign g1_0 = b[1:1];

  assign c$app_arg_0 = g1_0 ? 1'b1 : 1'b0;

  assign result_1 = c$app_arg;

  assign c$app_arg_1 = a[2:2] ? ({1'b1,(((g1_1)))}) : ({1'b0,({(c$app_arg_2),1'bx})});

  assign g1_1 = a[1:0];

  assign g1_2 = a[1:1];

  assign c$app_arg_2 = g1_2 ? 1'b1 : 1'b0;

  assign result_2 = c$app_arg_1;

  assign result = result_2 <= result_1;

endmodule
```
</details>

This just moves the processing into `pack`.

kleinreact, 2025-01-02:
> I do feel you're cheating here, I wonder if you even noticed.

Yeah! the point is that I'd like the generated code to be just `assign result = (a) <= (b);`, but that seems to be hard to archive with Clash, although it's the most straightforward solution.

DigitalBrains1, 2025-01-02:
I'd add custom primitives for your `BitPack` instance which are just the identity in HDL. Perhaps we could offer a custom `BitPack` and `Ord` derivation that does just that, with just a few keystrokes, for your custom data type.

That still leaves unsolved the fact that currently it's way too pessimistic (producing too much undefined values) for that, which is the main topic of this issue.

DigitalBrains1, 2025-01-02:
Ah, it's actually not going to work... the HDL simulation tools aren't clever enough.

<details><summary><code>OrdBV.hs</code></summary>

```haskell
module OrdBV where

import Clash.Explicit.Prelude
import Clash.Explicit.Testbench

data T = B Bool | D (Unsigned 2) deriving (Generic, BitPack, Eq, Ord)

topEntity :: BitVector (BitSize T) -> BitVector (BitSize T) -> Bool
topEntity a b = a <= b
-- topEntity a b = unpack @T a <= unpack b
{-# OPAQUE topEntity #-}

testBench ::
  Signal System Bool
testBench = done
 where
  inpVec :: Vec 64 (T, T)
  inpVec = map unpack $ iterateI (+1) 0
  (inpA, inpB) = unbundle $ stimuliGenerator clk rst inpVec
  expected = map (uncurry (<=)) inpVec
  out = topEntity <$> fmap pack inpA <*> fmap pack inpB
  done = outputVerifier' clk rst expected out
  clk = tbClockGen (not <$> done)
  rst = resetGen
{-# OPAQUE testBench #-}
```
</details>

<details><summary>Test suite results</summary>

```
.
  tests
    shouldwork
      OrdBV
        VHDL
          clash (gen):          OK (4.88s)
          tools
            GHDL
              import testBench: OK (0.03s)
              make testBench:   OK (2.50s)
              sim testBench:    OK (0.11s)
            Vivado
              sim testBench:    OK (31.54s)
        Verilog
          clash (gen):          OK (4.80s)
          tools
            IVerilog
              make testBench:   OK (0.14s)
              sim testBench:    FAIL (0.23s)
                Program /usr/bin/vvp (return code: 0) printed to stdout unexpectedly.
                
                Full invocation:
                
                  /usr/bin/vvp testBench.exe
                
                Stderr was:
                
                
                Stdout was:
                
                  @1000001ns: outputVerifier, expected: 1, actual: x
                Use -p '/OrdBV/&&/IVerilog.sim testBench/' to rerun this test only.
            Verilator
              make testBench:   OK (2.19s)
[... 133 lines cut ...]
                  # //
                  # Loading sv_std.std
                  # Loading work.OrdBV_testBench_types(fast)
                  # Loading work.testBench(fast)
                  # Loading work.topEntity(fast)
                  # 
                  # run -all
                  # @1000000ns: outputVerifier, expected: 1, actual: x
                  # ** Note: $stop    : /home/peter/tmp/clash-test_OrdBV-c76ee849c217fa85/modelsim-testBench/OrdBV.testBench/testBench.sv(1221)
                  #    Time: 100 ns  Iteration: 1  Instance: /testBench
                  # Break at /home/peter/tmp/clash-test_OrdBV-c76ee849c217fa85/modelsim-testBench/OrdBV.testBench/testBench.sv line 1221
                  # Stopped at /home/peter/tmp/clash-test_OrdBV-c76ee849c217fa85/modelsim-testBench/OrdBV.testBench/testBench.sv line 1221
                  # if {[string equal ready [runStatus]]} then {quit -f} else {quit -code 1 -f}
                  # End time: 12:55:37 on Jan 02,2025, Elapsed time: 0:00:02
                  # Errors: 0, Warnings: 1
                Use -p '/OrdBV/&&/ModelSim.sim testBench/' to rerun this test only.
            Verilator
              make testBench:   OK (3.29s)
              sim testBench:    OK

3 out of 17 tests failed (39.22s)
```
</details>

We can't outsmart the simulation tools, we can't produce definite values when the sim tool produces an indefinte value (other way around is fine).

DigitalBrains1, 2025-01-02:
I think we can close this because some simulation tools don't produce better results than what we have now, so there's no room for improvement.

(This disregards that your example is flawed, because the essence of producing more accurate results for `Ord BitPack` remains. But we can't.... :-(  ).

martijnbastiaan, 2025-01-02:
> I think we can close this because some simulation tools don't produce better results than what we have now, so there's no room for improvement.

Verilator even injects random bits whenever it sees an `X` --- though that's really a Verilator thing, not a Verilog thing.

martijnbastiaan, 2025-01-02:
@DigitalBrains1 For:

> I think we can close this because some simulation tools don't produce better results than what we have now, so there's no room for improvement.

If this is what the Verilog/VHDL spec prescribes then I'm 100% on board with this statement. If this is more an issue with simulators, wouldn't this just be an upstream issue? (Which we don't have to account for I think. (?))

DigitalBrains1, 2025-01-02:
> wouldn't this just be an upstream issue?

Yes, you can argue that. However, do we really want people's test benches to fail because of this? It'd be mighty hard to debug why your design suddenly starts producing X's somewhere. This is of course always something that might happen, but maybe we need to be a bit pragmatic here when basically all simulators we threw at it that support undefined bits seem to stumble on this (what's up with GHDL? I presume Verilator can be explained by it either not filling in random bits or the coin just landing correct side up for every equality test).

DigitalBrains1, 2025-01-02:
Hah! X hiding! VHDL `boolean` doesn't support undefined values, and apparently it happens to give the correct result then? Probably by chance.

kleinreact, 2025-01-02:
The Verilog standard states

_An expression using these relational operators shall yield the scalar value 0 if the specified relation is false or the
value 1 if it is true. If, due to unknown or high-impedance bits in the operands, the relation is ambiguous, then the
result shall be a 1-bit unknown value (x)._

Hence, as @DigitalBrains1 already noted, the `<=` relation indeed is ambiguous because of the equality part, but `<` wouldn't be ambiguous here. 

@DigitalBrains1 can you retry your experiment with `<` instead of `<=`? Does it yield the same result?

martijnbastiaan, 2025-01-02:
Related discussion: https://github.com/clash-lang/clash-compiler/issues/2194

DigitalBrains1, 2025-01-02:
Sure! Results are the same.

<details><summary>Test suite results</summary>

```
.
  tests
    shouldwork
      OrdBV
        VHDL
          clash (gen):          OK (4.86s)
          tools
            GHDL
              import testBench: OK (0.04s)
              make testBench:   OK (2.64s)
              sim testBench:    OK (0.12s)
            Vivado
              sim testBench:    OK (38.94s)
        Verilog
          clash (gen):          OK (4.79s)
          tools
            IVerilog
              make testBench:   OK (0.17s)
              sim testBench:    FAIL (0.24s)
                Program /usr/bin/vvp (return code: 0) printed to stdout unexpectedly.
                
                Full invocation:
                
                  /usr/bin/vvp testBench.exe
                
                Stderr was:
                
                
                Stdout was:
                
                  @1000001ns: outputVerifier, expected: 0, actual: x
                Use -p '/OrdBV/&&/IVerilog.sim testBench/' to rerun this test only.
            Verilator
              make testBench:   OK (2.36s)
              sim testBench:    OK
            Vivado
              sim testBench:    FAIL (42.07s)
                Program /home/peter/Xilinx/Vivado/2022.1/bin/vivado (return code: 0) printed to stdout 
                unexpectedly.
                
                Full invocation:
                
                  /home/peter/Xilinx/Vivado/2022.1/bin/vivado -mode batch -source /home/peter/tmp/clash-test_OrdBV-053bd21d56249366/vivado-testBench/sim.tcl
                  
                
                Stderr was:
                
                
                Stdout was:
                
                  
                  ****** Vivado v2022.1.1 (64-bit)
                    **** SW Build 3557992 on Fri Jun  3 09:56:20 MDT 2022
                    **** IP Build 3556342 on Fri Jun  3 19:31:36 MDT 2022
                      ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
                  
                  INFO: [Common 17-1239] XILINX_LOCAL_USER_DATA is set to 'no'.
                  Sourcing tcl script '/home/peter/.Xilinx/Vivado/Vivado_init.tcl'
                  source /home/peter/tmp/clash-test_OrdBV-053bd21d56249366/vivado-testBench/sim.tcl
                  # set_msg_config -severity {CRITICAL WARNING} -new_severity ERROR
                  # source -notrace {/home/peter/src/clash/clash-compiler/./clash-lib/./data-files/tcl/clashConnector.tcl}
                  # clash::readMetadata {/home/peter/tmp/clash-test_OrdBV-053bd21d56249366/vivado-testBench/OrdBV.testBench/}
                  Clash [1]: New top component: testBench
                  Clash [1]: New top component: topEntity
                  Clash [1]: Top entity is testBench
[... 102 lines cut ...]
                  # //
                  # Loading sv_std.std
                  # Loading work.OrdBV_testBench_types(fast)
                  # Loading work.testBench(fast)
                  # Loading work.topEntity(fast)
                  # 
                  # run -all
                  # @1000000ns: outputVerifier, expected: 0, actual: x
                  # ** Note: $stop    : /home/peter/tmp/clash-test_OrdBV-8b55ec7055b9179b/modelsim-testBench/OrdBV.testBench/testBench.sv(1215)
                  #    Time: 100 ns  Iteration: 1  Instance: /testBench
                  # Break at /home/peter/tmp/clash-test_OrdBV-8b55ec7055b9179b/modelsim-testBench/OrdBV.testBench/testBench.sv line 1215
                  # Stopped at /home/peter/tmp/clash-test_OrdBV-8b55ec7055b9179b/modelsim-testBench/OrdBV.testBench/testBench.sv line 1215
                  # if {[string equal ready [runStatus]]} then {quit -f} else {quit -code 1 -f}
                  # End time: 13:50:48 on Jan 02,2025, Elapsed time: 0:00:01
                  # Errors: 0, Warnings: 1
                Use -p '/OrdBV/&&/ModelSim.sim testBench/' to rerun this test only.
            Verilator
              make testBench:   OK (3.57s)
              sim testBench:    OK

3 out of 17 tests failed (46.88s)
```
</details>

[image]

DigitalBrains1, 2025-01-02:
> Hence, as @DigitalBrains1 already noted, the <= relation indeed is ambiguous because of the equality part, but < wouldn't be ambiguous here.

That still seems wrong... same case. `B False < B False = False`, but `0b0. < 0b0.` is undefined. It's still that you can't do equality, you've just inverted the result for equality.

kleinreact, 2025-01-02:
Thanks for the discussion and insights. I close this, as I wasn't aware that the standards basically don't allow interpretation of partially defined inputs. Comparison via the packed values thus is not very useful in the scope of Clash.

DigitalBrains1, 2025-01-02:
I feel the standard allows you to do this. The HDL simulators miss an opportunity here. I'm not really comfortable with outsmarting the simulators for reasons already stated.

JvWesterveld, 2025-01-02:
> Hah! X hiding! VHDL `boolean` doesn't support undefined values, and apparently it happens to give the correct result then? Probably by chance. [...]

Yes, in VHDL `ieee.numeric_std."<="` gives a note of severity warning when encountering metavalues.
https://gitlab.com/IEEE-P1076/packages/-/blob/f181327f309acc602d5d1899e261e12ec61e6ee7/ieee/numeric_std-body.vhdl#L1454

kleinreact, 2025-01-02:
>I feel the standard allows you to do this. The HDL simulators miss an opportunity here. I'm not really comfortable with outsmarting the simulators for reasons already stated.

At least with the 2001 standard I don't see any freedom here: if any bit of the input is undefined or z then the output must be undefined!

> But for many ADT's T, Ord T simply doesn't match Ord BitVector under BitPack T.

Do you have a concrete example? As far as I recognized, they are always compatible.

DigitalBrains1, 2025-01-02:
> At least with the 2001 standard I don't see any freedom here:

Agreed. Apparently I just read the quote from 1995 somehow. It's weird how you sometimes realise you completely missed a paragraph in a written discussion.

> Do you have a concrete example? As far as I recognized, they are always compatible.

Your `T` right here. `a <= b` does not match `pack a <= pack b` when `a = b`. Neither does any other `Ord` function.

kleinreact, 2025-01-02:
Ah right. I just was thinking compatibility the other way around:

_If `pack a <= pack b` is unambiguous with respect to the undefined bits, and therefore could return a defined result, then `a <= b` would be compatible with that result._

DigitalBrains1, 2025-01-02:
I don't see how that helps you write a more simple `Ord T` though.

kleinreact, 2025-01-02:
> I don't see how that helps you write a more simple Ord T though.

It doesn't! I just missed the equality case and that I it even hurts you when using `<`. Hence, again thanks for pointing that out.

DigitalBrains1, 2025-01-02:
> Yes, in VHDL `ieee.numeric_std."<="` gives a note of severity warning when encountering metavalues. https://gitlab.com/IEEE-P1076/packages/-/blob/f181327f309acc602d5d1899e261e12ec61e6ee7/ieee/numeric_std-body.vhdl#L1454

That doesn't match what I'm seeing in Vivado. Note how at 100 ns and 280 ns it is returning TRUE not FALSE. There are also no warnings in the Tcl Console.

DigitalBrains1, 2025-01-02:
So my colleague Just dove a bit deeper into that VHDL code and it is in fact correct: the `TO_01` function outputs undefined bits on every position if its inputs contain any undefined bits. However, the Vivado sim is still flawed. It actually does some flawed comparison on undefined bits. The following is `a <= b` in VHDL (with the output converted to `std_logic_vector` because that's what I wanted in the Clash code). You can see that if you swap the operands, the results also change, clearly showing it is somehow incorrectly trying to compare undefined bits to defined bits.

[image]

I wonder if GHDL does the same. I should try one of these days.

leonschoorl, 2025-01-06:
@kleinreact 
> If `pack a <= pack b` is unambiguous with respect to the undefined bits, and therefore could return a defined result, then `a <= b` would be compatible with that result.

[Down](https://hackage.haskell.org/package/base-4.14.3.0/docs/Data-Ord.html#t:Down) explicitly flips the Ord instance thus making it incompatible:
```
clashi> import Data.Ord
clashi> a = 3 :: Down Int
clashi> b = 4 :: Down Int
clashi> a <= b
False
clashi> pack a <= pack b
True
```

DigitalBrains1, 2025-01-06:
I implicitly read it as "a user type with the standard `deriving (BitPack, Ord)`". A custom `BitEncoding` would be another example where they sort in a different order.