clash-lang/clash-compiler #2867: 1.8.2 doesn't have a Counter instance for Maybe a

gergoerdi, 2025-01-04:
Is there a good reason 1.8.2 doesn't contain the `Counter a => Counter (Maybe a)` instance? I use that in my Sudoku solver and my stream formatter and would love to use a stable release of Clash for a paper I'm writing on them.

christiaanb, 2025-01-04:
It seems https://github.com/clash-lang/clash-compiler/pull/2692 was not labeled as `backport-1.8`?

christiaanb, 2025-01-04:
I guess it could have had a `backport-1.8` label if I look at the [PVP](https://pvp.haskell.org/):

> Non-breaking change. Otherwise, if only new bindings, types, classes, non-orphan instances or modules (but see below) were added to the interface, then A.B MAY remain the same but the new C MUST be greater than the old C. Note that modifying imports or depending on a newer version of another package may cause extra non-orphan instances to be exported and thus force a minor version change.

DigitalBrains1, 2025-01-05:
My first instinct is to only backport fixes for something (not just bugs, also documentation and more stuff). Ideally I suppose we'd raise the issue ourselves on external PR's and ask "how soon would you like to have this", but Cat knows I'm not ideal! So it definitely helps if you give us a nudge in the right direction and ask for a backport!

We can just do another 1.8.3 soon. It's not like we'd run out of numbers.