clash-lang/clash-compiler #2875: Seemingly innocent function takes very long to synthesize

rowanG077, 2025-01-15:
The following:

```Haskell
foo
  :: forall
    (deadlinePicoSeconds :: Nat)
    (dom :: Domain)
   . (HiddenClockResetEnable dom)
  => (KnownNat (DomainPeriod dom))
  => (1 <= DomainPeriod dom)
  => (KnownNat deadlinePicoSeconds)
  => 1 <= DivRU deadlinePicoSeconds (DomainPeriod dom)
  => Signal dom Bool
  -> Signal dom Bool
foo activity =
  let initCnt = (0 :: Index (DivRU deadlinePicoSeconds (DomainPeriod dom)))
      cntr = register initCnt $ mux activity (pure 0) (satSucc SatWrap <$> cntr)
  in register False $ cntr .==. (pure maxBound)

topEntity = exposeClockResetEnable (foo @5_000_000_000_000 @System)
```

Takes a whopping 25 seconds to synthesize on my machine using GHC 9.6.6 and clash 1.8.2. As part of a larger circuit it never finished synthesizing.

rowanG077, 2025-01-15:
Normalization is not to blame, it's stuck at the `Clash: Compiling ...` stage.

christiaanb, 2025-01-15:
Does compiling the file with regular GHC to an `.o` object file also take longer when you use GHC 9.6.6 vs 9.0.2?

rowanG077, 2025-01-15:
Doesn't seem to be different. How large `deadlinePicoSeconds` is also doesn't factor into it. With both it's around 0.150 seconds.

Just to be clear. It takes so long after clash normalization has taken place. This is the output:

```
GHC: Setting up GHC took: 0.040s
GHC: Compiling and loading modules took: 0.078s
Clash: Parsing and compiling primitives took 0.133s
GHC+Clash: Loading modules cumulatively took 0.497s
Clash: Compiling TopEntity.topEntity
Clash: Normalization took 0.012s
Clash: Netlist generation took 0.000s
-- Here it takes a ages
Clash: Compiling TopEntity.topEntity took 23.663s
Clash: Total compilation took 24.161s
```

rowanG077, 2025-01-15:
A smaller reproducer::

```
foo
  :: forall
    (deadlinePicoSeconds :: Nat)
    (dom :: Domain)
   . (HiddenClockResetEnable dom)
  => (KnownNat (DomainPeriod dom))
  => KnownNat (DivRU deadlinePicoSeconds (DomainPeriod dom))
  => (1 <= DomainPeriod dom)
  => (KnownNat deadlinePicoSeconds)
  => 1 <= DivRU deadlinePicoSeconds (DomainPeriod dom)
  => Signal dom (Index (DivRU deadlinePicoSeconds (DomainPeriod dom)))
foo = register 0 foo

topEntity = exposeClockResetEnable (foo @5_000_000_000_000 @System)
```

Here the synthesis time is not quite as bad. But still pretty bad and still linear in `deadlinePicoSeconds`.

rowanG077, 2025-01-15:
It's actually just literally how large the `n` is for an `Index`...

```Haskell
topEntity :: Index 1_000_000_000
topEntity = 0
```

Takes 5 seconds on my machine.

```Haskell
topEntity :: Index 5_000_000_000
topEntity = 0
```

Takes approx 27 seconds. Amazing.

christiaanb, 2025-01-16:
Probably a duplicate of https://github.com/clash-lang/clash-compiler/issues/2813, in which case it will be fixed by https://github.com/clash-lang/clash-compiler/pull/2816

rowanG077, 2025-01-16:
Yes this seems to be the case. I will close this.

closed 2025-01-16 by rowanG077