clash-lang/clash-compiler #2888: `Enum` instance of `UFixed` is not Haskell conform

kleinreact, 2025-02-13:
As already noted [here](https://github.com/clash-lang/clash-compiler/pull/2858#issuecomment-2656193232), the `Enum` instances of `UFixed` is not `Haskell2010` conform: 
_(the arguments of the prior discussion are restated below for convenience)_

--- 

I consider the successor of `1.5 :: UFixed 2 1` to be `2`and not `2.5`.

The current `Enum (UFixed 2 1)` instance implements a similar behavior as the `Enum` instance for `Float`, but the report makes an explicit exception for that particular type. It defines:

_For all **four** numeric types_ **(which are only `Int`, `Integer`, `Float`, and `Double`)** _, succ adds 1, and pred subtracts 1_.

For **any** other type the semantics is:

_Class `Enum` deﬁnes operations on sequentially ordered types. The functions succ and pred return the
successor and predecessor, respectively, of a value._

The sequential order of `UFixed 2 1` is
```
0.0, 0.5, 1.0, 1.5, 2.0, 2.5, 3.0, 3.5
```
and, thus, the successor of `1.5` must be `2.0`.

As `UFixed 2 1` also has a `Bounded` instance, there even is proof that the current implementation is wrong, because in that case it is also required that

```haskell
enumFromThen x y = enumFromThenTo x y bound
  where
    bound | fromEnum y >= fromEnum x = maxBound
          | otherwise = minBound
```
which does not hold for `UFixed 2 1`, `x = 2.5` and `y = 2.0`, resulting in
```
[2.5,2.0,1.5,1.0,0.5,0.0] = []
```

martijnbastiaan, 2025-02-13:
I get what you're saying, but I think a literal reading of the `Enum` class laws is pretty useless. These laws were written almost 30 years ago, back when there were very few numeric types to start with. From the mentioned four, only two extra existed: `Ratio` and `Complex`. The former doesn't have a `Bounded` instance (and its `succ` behaves like `+1`), and the latter doesn't have an `Enum` instance.

And so, if anything, we should read the carve out for `Int`..`Double` as a sign that the authors did realize just how silly the laws are for numeric types, that they didn't foresee `Bounded` numeric types, or both. In fact you'll see that the documentation of `Enum` just considers _numeric_ types not _the four numeric types_. E.g., for `succ`:

> Successor of a value. For numeric types, [`succ`](https://hackage.haskell.org/package/base-4.21.0.0/docs/Prelude.html#v:succ) adds 1.

This all leads me to the conclusion that in 2025 `Enum` should just be considered a way of getting `[a, b ..]` to do something non-silly. It's a hackjob of a class (*cough* `fromEnum :: a -> Int`, partial functions *cough*), but not quite terrible enough to warrant major breaking changes.

On top of that, as Peter mentioned in the other thread, we already changed the `Enum` a while ago to work with the interpretation of `Enum` I just laid out. This has caused at least some users a great deal of debug pain. So even if you think that reasoning is all wrong, it would be a great disservice to our users to change it _again_, only to potentially flip-flop back again years later.

To wrap it up: I don't think this is a bug on our end, nor would it be an improvement to abide by the mentioned laws. The only feasible way to improve on the situation would be to offer our own class, which I feel `Finite` was on its way of becoming.

christiaanb, 2025-02-14:
`clash-prelude` is not alone in that it adds `1` to its Fixed number types:
* https://github.com/ekmett/fixed/blob/b11f57ba89da99e0e5cc41a9af00da5845606ce5/src/Numeric/Fixed.hs#L54-L55
* https://github.com/cartazio/numbers/blob/0ab9e067e10a2e097239f73bdcbbffeae09790b0/Data/Number/Fixed.hs#L59

DigitalBrains1, 2025-02-16:
> On the other hand, we are deviating in behavior from the base implementation of [Fixed](https://hackage.haskell.org/package/base-4.21.0.0/docs/Data-Fixed.html#t:Fixed). Why?

For the really basic reason that when we implemented the current behaviour of our `Fixed`, the `Enum` instance for `Data.Fixed` was undocumented: https://hackage.haskell.org/package/base-4.14.0.0/docs/Data-Fixed.html#t:Fixed

DigitalBrains1, 2025-02-16:
Finally, people will often make a design with `Float` and then switch to our `Fixed` once they have something that works in simulation and it is time to work towards a synthesizable design. So I'd rather not deviate from `Float` if not necessary. And `Float` also has finitely many inhabitants (obviously, since it's fixed in size), so an `Enum` instance could have enumerated them just fine and you'd get the same behaviour: `succ` would give the next larger representable value, and `pred` the next smaller representable value. Yet they chose not to do this.

christiaanb, 2025-02-16:
Actually, you can’t make `Enum Float` behave exactly like `Enum Data.Fixed`. `Enum Data.Fixed` has:

1.  a `succ` and `pred` with the same fraction every time, and 
2. where `a /= succ a` and `a /= pred a` hold

For `Enum Float` to have 2, you would need to increment/decrement the mantissa by one; but than you wouldn’t have 1 (because the exponent determines the size of the fraction). The existing `Enum Float` option picked aspect 1, but then you see that it does not have aspect 2 when you have a large number.

The fraction that they ended up choosing, `1`, actually feels really arbitrary to me; where the only thing it had going for it is that is was like `Int` and `Integer`.  But I think incrementing/decrementing the mantissa actually would make more sense, so that `succ 3e38 :: Float` wouldn’t be a NOP like it is now.

And when it comes to `Enum Rational` picking a smallest fraction wouldn’t make sense because it’s infinite precision. So while picking `1` as the faction to increment/decrement is arbitrary; at least it satisfies both aspect 1 and 2 I mentioned earlier. Since all choices for the fraction would’ve been equally arbitrary, keeping it the same as `Int` and `Integer` is the least surprising one.

When we first changed the definition of `Enum Clash.Sized.Fixed` to use `1` as the fraction we increment and decrement with I really didn’t give it much thought; and just accepted the class documentation. However, now that

1. I think that the language designers had to make an arbitrary choice for `Enum Rational` and `Enum Float` as to what to pick as the fraction to increment/decrement with; where I think they choose wrongly for `Enum Float`
2. That `Enum Data.Fixed` picks the smallest fraction to increment/decrement with since its inception in `base-2`

I would be in favor to make `Enum Clash.Sized.Fixed` again also pick the smallest possible fraction to increment/decrement in its `succ` and `pred` implementation. I think I would ask the users who suffered from the fallout of switching `Enum Clash.Sized.Fixed` to increment/decrement with `1` how they feel about it though.

DigitalBrains1, 2025-02-17:
> Actually, you can’t make `Enum Float` behave exactly like `Enum Data.Fixed`.

I never intended to claim that. I intended to claim that you can actually make `succ` pick the next larger representable value and `pred` pick the next smaller representable value, and have them enumerate the full space of inhabitants of the type. Although you'd have to think about negative zero, since both `Eq` and `Ord` consider negative zero and zero to have the same value/ordering (and remember that pattern matches on numbers are expressed in `Eq`, so distinghuishing the two zeros in code needs more than usual). And I feel by default you wouldn't enumerate any of the 2^24-2 `NaN`s just like `Finite` by default doesn't enumerate undefined bits in `BitVector`. So it wouldn't actually enumerate the full space of inhabitants probably, by a strict definition.

> When we first changed the definition of `Enum Clash.Sized.Fixed` to use `1` as the fraction we increment and decrement with I really didn’t give it much thought; and just accepted the class documentation.

I remember quite extensive discussions, but apparently you were not a part of them then :-D. We were aware neither choice was the obviously correct one, but that it would be surprising to deviate from the class documentation.

> I would be in favor to make `Enum Clash.Sized.Fixed` again also pick the smallest possible fraction to increment/decrement in its `succ` and `pred` implementation.

I think I remembered my strongest argument and I already mentioned it above: people might very well start with `Float` to make a design that works in simulation and then start replacing those `Float`s by `Fixed`s once they're in the phase of turning their design into a synthesizable design. This means that if they used `succ` and `pred` in their design with `Float`s it suddenly behaves differenly when they replace a `Float` by a `Fixed`.

Additionally, I think the behaviour of picking the next representable value is a useful behaviour to have, but `succ` and `pred` don't need to be the functions that provide this behaviour. Currently it's a bit cumbersome because of the type information lost by a function like [`unUF`](https://hackage.haskell.org/package/clash-prelude-1.8.2/docs/Clash-Sized-Fixed.html#v:unUF) which means that `uf . succ . unUF` might need a type signature. But the functions `finSucc` and `finPred` of the `Finite` class would be a perfect candidate for this functionality.

The functionality of _I want to add 1 or subtract 1 regardless of the type_ is provided by `satPred` and `satSucc`, which is also a useful behaviour to have. By the way, I feel it would be weird if `satPred` and `satSucc` differed much from `pred` and `succ`. This is fine for alternative interpretations of enumeration, but those `sat` ones were specifically designed to mimic `pred` and `succ` while providing more useful boundary behaviour.

I see no need to horse around with `succ` and `pred` again...

kleinreact, 2025-02-17:
I think the general misconception here is to assume that `succ` and `pred` have the semantic meaning of adding or subtracting `1`. According to the report, the semantic meaning of `succ` and `pred` is to pick the next representable value, where the implementation for `Float` is just exceptional, because picking such a value there isn't well defined. If you want to semantically implement "add 1" than you simply should use `(+ 1)` instead.

The names `satPred` and `satSucc` are also misleading in that sense, because I'd interpret them (just from the names) as "picking the next / previous representable + chosen wrapping behavior", which is in conflict with the documentation _"Get successor / predecessor of (or in other words, add 1 to) given number"_ of our documentation.

@DigitalBrains1 I have a concrete question: Why do you consider `succ` and `pred` to have the semantics of adding 1? Just because base's Haddock documentation is ambiguous in that regard? Because the _Haskell Report_ is not.

DigitalBrains1, 2025-02-17:
As I've indicated before, we don't agree on our interpretation of `Enum` laws and rules. Let's agree to disagree because we've discussed this enough now. You don't have to keep repeating your interpretation.

> If you want to semantically implement "add 1" than you simply should use (+ 1) instead.

I've said it several times before: this doesn't work for degenerate cases.

```haskell
clashi> minBound :: SFixed 0 2
-0.5
clashi> maxBound :: SFixed 0 2
0.25
clashi> minBound + 1 :: SFixed 0 2
-0.25
clashi> satSucc SatError minBound :: SFixed 0 2
*** Exception: X: Fixed.satSucc: result exceeds maxBound
CallStack (from HasCallStack):
  errorX, called at src/Clash/Sized/Fixed.hs:1153:29 in clash-prelude-1.8.2-KDigOd4FFBnLPtX7qtI31r:Clash.Sized.Fixed
clashi> satSucc SatBound minBound :: SFixed 0 2
0.25
```

DigitalBrains1, 2025-02-17:
Felix, did you accidentally skip reading [Martijns message](https://github.com/clash-lang/clash-compiler/issues/2888#issuecomment-2657763637) above?

kleinreact, 2025-03-09:
> Felix, did you accidentally skip reading [Martijns message](https://github.com/clash-lang/clash-compiler/issues/2888#issuecomment-2657763637) above?

No, I read that message very carefully and consider it to be a very important and valuable PoV, although not matching with mine.

---

**I probably should have stated this more clearly upfront:** my intention of creating this issue is not to declare something to be wrong or a bug, but just to make people aware that there are some diverging PoVs which might lead to confusion and problems. This discussion very clearly shows that there are different perspectives and that there is no right or wrong. 

Not having a clear consensus about what is right or wrong is an issue, because it might lead to confusion and that again might lead to bugs or wrong behavior in the end.

Our goal should be to simply collect all these technical PoVs and to make it clear for the user which ones are implemented by Clash and which ones are not. In particular, this clarification should not only happen in the discussion here, but also should be part of the documentation of our code.

---

Hence, regarding the collection of technical arguments, I like to add one more bit to the table, because it hasn't been brought up yet and I still consider it to be valuable:

The documentation of `base` has a [section](https://hackage.haskell.org/package/base-4.21.0.0/docs/Prelude.html#g:6) explicitly enumerating the `Numeric Types`, which currently are `Int`, `Integer`, `Float`, `Double`, `Rational` and `Word`. I found this, because I was looking for a definition of _"what actually is a **numeric type**"_ and this was the best I could find. If you know about any better definition, I'm happy to learn about that. 

On a final regard, I consider it to be intentional that `Fixed` is not listed here, but we do not have or need to agree on that.