clash-lang/clash-compiler #2895: Issue with `KnownNat` when synthesizing

diegodiv, 2025-02-21:
Something happened to me the other day, and it seems the culprit was one of my uses of `KnownNat`.

Here is a minimal reproducer (thanks @kleinreact):
```haskell
import Clash.Prelude

topEntity :: KnownNat 19 => Signed 5
topEntity = go0 SNat

go0 :: SNat 19 -> Signed 5
go0 SNat = resize $ go1 d19

go1 :: SNat n -> Signed 19
go1 _ = undefined
```
It compiles just fine, but gives rise to this error on synthesis:
```
Clash.Netlist.BlackBox(212): Couldn't instantiate blackbox for Clash.Sized.Internal.Signed.resize#. Verification procedure reported: 
                                                                                                                                     
Argument 0 should be literal, as blackbox used ~LIT[0], but was:                                                                     
                                                                                                                                     
Identifier (RawIdentifier "_$dKnownNat" Nothing []) Nothing                                                                          
                                                                                                                                     
The source location of the error is not exact, only indicative, as it is acquired                                                    
after optimizations. The actual location of the error can be in a function that is                                                   
inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.                                               
                                                                                                                                     
 topEntity = go0 SNat                                                                                                                
```
Removing the `KnownNat 19` constraint on `topEntity` makes it synthesizeable.

kleinreact, 2025-02-21:
With the `debug` flag enabled it gives a bit more insights, as tested with the current `master`:
```
Clash.Netlist.BlackBox(212): Couldn't instantiate blackbox for Clash.Sized.Internal.Signed.resize#. Verification procedure reported:

Argument 0 should be literal, as blackbox used ~LIT[0], but was:

Identifier (RawIdentifier "_$dKnownNat" Nothing [("unsafeMake",SrcLoc {srcLocPackage = "clash-lib-1.9.0-inplace", srcLocModule = "Clash.Netlist.Id", srcLocFile = "src/Clash/Netlist/Id.hs", srcLocStartLine = 244, srcLocStartCol = 20, srcLocEndLine = 244, srcLocEndCol = 30}),("unsafeFromCoreId",SrcLoc {srcLocPackage = "clash-lib-1.9.0-inplace", srcLocModule = "Clash.Netlist.BlackBox", srcLocFile = "src/Clash/Netlist/BlackBox.hs", srcLocStartLine = 287, srcLocStartCol = 32, srcLocEndLine = 287, srcLocEndCol = 51})]) Nothing

The source location of the error is not exact, only indicative, as it is acquired
after optimizations. The actual location of the error can be in a function that is
inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.

topEntity = go0 SNat
```

rowanG077, 2025-02-21:
I didn't look at the core but I expect this is a duplicate of https://github.com/clash-lang/clash-compiler/issues/1028

christiaanb, 2025-02-21:
I guess the work-around for this issue is to use `~SIZE[~TYP[2]]` instead of `~LIT[0]` to get an integer literal corresponding to the `n` in the `Signed n` argument in the blackboxes for the `resize#` function.

leonschoorl, 2025-02-21:
Looking at the core this seems very hard to solve:
with `topEntity :: KnownNat 19 => Signed 5`
```
topEntity before normalization:

λ($dKnownNat :: KnownNat 19) ->
resize# @5 @19 $dKnownNat[LocalId] (NS 5##) (undefined[GlobalId] @(Signed 19) go2[GlobalId])
```

vs with `topEntity :: Signed 5`
```
topEntity before normalization:

resize# @5 @19 (NS 19##) (NS 5##) (undefined[GlobalId] @(Signed 19) go2[GlobalId])
```

@christiaanb's workaround is very clever, and seems like it should work in this case.

But I think in general you should avoid adding superfluous constraints.
But especially to synthesis boundaries.
Maybe clash could even be made to throw warnings/errors about them.

As it very easily run you into problems:
```
topEntity :: Eq Int => Unsigned 5
topEntity = 3
```
Seems a little pointless maybe, but harmless enough in Haskell.

But what that type signature is actually saying is, my topEntity has an input port of type:
`Eq (Int -> Int -> Bool) (Int -> Int -> Bool)`
Even though it's unused, it is there in the definition.

And clash attempt to calculate a hardware type for that input port containing function types:
```
error:    
    Clash.Netlist.Util(784): Can't translate non-tycon type: Int -> Int -> Bool
    
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
   |
 4 | topEntity = 3
```

martijnbastiaan, 2025-02-21:
For newer GHCs we should be able to see it's a dictionary. We could conjure them up for synthesis boundaries (as suggested in the issue Rowan links), constant propagate, and remove it. This will still get you into trouble if you do `Eq SomeTypeWithoutAnEqInstance` though..

I guess the simplest and most foolproof way would just be to reject dictionaries on synthesis boundaries.