clash-lang/clash-compiler #2929: Compiler hangs on SNat computation

Rowan-Mather, 2025-04-09:
**The Problem:**
When compiling code similar to the example below, the compiler gets as far as something like `Clash: Compiling topEntity`, and then hangs.

I originally compiled it with Clash 1.6.5 and GHC 8.10.7, but I believe the problem persists to Clash 1.8.2.

**Example Notes:**
When we set `expSizeInt :: Int`, we see the compiler hang, but when we set `expSizeInt :: Unsigned 4`, it completes. I would therefore suggest it could be a bug in the specific treatment of Ints.

**Example:**
```
{-# LANGUAGE AllowAmbiguousTypes #-}
{-# LANGUAGE BlockArguments #-}
{-# LANGUAGE DataKinds #-}
{-# LANGUAGE FlexibleContexts #-}
{-# LANGUAGE GADTs #-}
{-# LANGUAGE MultiParamTypeClasses #-}
{-# LANGUAGE NoImplicitPrelude #-}
{-# LANGUAGE NoStarIsType #-}
{-# LANGUAGE NoTypeInType #-}
{-# LANGUAGE RankNTypes #-}
{-# LANGUAGE ScopedTypeVariables #-}
{-# LANGUAGE TypeApplications #-}
{-# LANGUAGE TypeFamilies #-}
{-# LANGUAGE TypeOperators #-}
{-# LANGUAGE UndecidableInstances #-}
{-# LANGUAGE DerivingStrategies #-}
{-# LANGUAGE DeriveGeneric #-}

{-# OPTIONS_GHC -fplugin CircuitNotation              #-}
{-# OPTIONS_GHC -fplugin GHC.TypeLits.Extra.Solver    #-}
{-# OPTIONS_GHC -fplugin GHC.TypeLits.KnownNat.Solver #-}
{-# OPTIONS_GHC -fplugin GHC.TypeLits.Normalise       #-}
{-# OPTIONS_GHC -fconstraint-solver-iterations=20     #-}

module HangingCompilationExample where

-- base
import qualified Prelude as P

-- clash-prelude
import Clash.Prelude

brainToIndexSimple
  :: forall expSize n k
   . (1 <= expSize
     , expSize <= 8
     , KnownNat n
     -- k constraints are to assist constraint checker
     , KnownNat k
     , k ~ (1 + expSize + n)
     , CLog 2 (2 ^ k) ~ k
     )
   => SNat expSize
   -> Index (2 ^ (1 + expSize + n))
brainToIndexSimple SNat
  = let
  -- If we replace the type with Unsigned 4 here, it should compile
  expSizeInt :: Int -- Unsigned 4
  expSizeInt = snatToNum (SNat @expSize)
  expSat :: Unsigned expSize
  expSat = 2 P.^ expSizeInt - 2
  in unpack $ resize $ pack expSat

topEntity :: Index 65536
topEntity = brainToIndexSimple d8
```

Rowan-Mather, 2025-04-10:
Upon further testing, it seems this is fixed when using GHC 9.6.2 and Clash `master`.

DigitalBrains1, 2025-04-15:
Hello, sorry that it took a while before we got back to you!

I can reproduce this in Clash 1.6.5, but not in 1.8.2. Is this really still a problem in 1.8.2? Does this exact reproducer cause issues for you? If so, can you tell me more about your setup, give me hints on how to get something close to what you have?

Rowan-Mather, 2025-04-16:
Hey, thanks for your response! My apologies for misleading you, I hadn't updated clash properly when I quickly checked 1.8.2. I've now gone through and verified it hangs in 1.6.5 but not 1.8.2 (both with GHC 8.10.7). 

So there isn't necessarily a bug still to be fixed here. But it would be interesting to know what the change was, so the bug is easier to spot when we have to use this older version. I'll see what I can find...

DigitalBrains1, 2025-04-16:
> My apologies for misleading you, I hadn't updated clash properly

No worries, these things happen :-).

> But it would be interesting to know what the change was, so the bug is easier to spot when we have to use this older version.

Ah, please do note that we no longer support 1.6.5. We know it contains bugs, but the fix is upgrading to 1.8. Currently, only 1.8 is the supported version.

That said, if upgrading your code to be 1.8-compatible is intractable for now, we might be able to offer some assistance if it's not much work. Perhaps you can bisect to find the git commit introducing the fix, and then apply that commit to your local 1.6.5 install. Or if you don't know how to, I might be able to, time permitting. I'm not making any promises, after all, we don't support 1.6 at all anymore.

closed 2025-10-14 by Rowan-Mather