clash-lang/clash-compiler #2949: Poor errors/documentation for nested data types with custom bit representations

kleinreact, 2025-05-02:
I'm not fully aware of how the framework is intended to be used, but the current documentation and the printed error messages don't give much insight.

If I try to GHC compile

```haskell
data T = T0 Bool
{-# ANN module (DataReprAnn $(liftQ [t|T|]) 1
  [ConstrRepr 'T0 0b0 0b0 [0b0]]) #-}

deriveBitPack [t|T|]
```

then I get

```
    Unexpected empty list of intervals
   |
12 | deriveBitPack [t|T|]
   | ^^^^^^^^^
```
via trial and error I figured out that updating the last `[0b0]` argument of `ConstrRepr` to `[0b1]` makes `deriveBitPack` happy, but now `clash` gives me

```
<no location info>: error:
    Clash error call:
    Unexpected HWType: Bool
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/Util.hs:316:10 in clash-lib-1.9.0-inplace:Clash.Netlist.Util
      convertToCustomRepr, called at src/Clash/Netlist/Util.hs:356:8 in clash-lib-1.9.0-inplace:Clash.Netlist.Util
```

I don't get any feedback of whether I just try to do something stupid here, or whether there is an error in with my annotation. The current documentation and the given error messages are not really helpful at this point.

DigitalBrains1, 2025-05-02:
Yes, this is definitely more magic and unclear than you'd want.

> <code>ConstrRepr 'T0 0b0 0b0 _**[0b0]**_</code>

The documentation does _try_ to explain that that final list is a list of bitmasks detailing in which bits to store every field, but it can definitely be improved. What you're saying here is that the first and only field, the `Bool`, needs to be encoded in 0 bits. As you discovered, you need to say it is encoded in 1 bit, the sole bit of the encoded data.

Regarding
```
Unexpected HWType: Bool
```
I think you've discovered a bug that nobody was aware of because nobody tried to make a custom repr for a single-constructor custom data type before. Because if I make it something with two constructors, everything seems fine:
```haskell
data T = T0 Bool | T1
{-# ANN module (
  DataReprAnn $(liftQ [t|T|]) 2
    [ ConstrRepr 'T0 0b10 0b00 [0b01]
    , ConstrRepr 'T1 0b10 0b10 []
    ]) #-}
```

So let's make an issue for that as well, since it is separate from this issue.

kleinreact, 2025-05-03:
Another example
```haskell
data T where
  T0 :: Bool -> Bit -> T

{-# ANN module (DataReprAnn $(liftQ [t|T|]) 2
  [ ConstrRepr 'T0 0b00 0b00 [0b10]
  ]) #-}
```
gives you
```
   Clash error call:
    Clash.Netlist(1028): Over-applied constructor: Check.T0
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist.hs:1028:17 in clash-lib-1.9.0-inplace:Clash.Netlist
      mkDcApplication, called at src/Clash/Netlist.hs:835:16 in clash-lib-1.9.0-inplace:Clash.Netlist
      mkExpr, called at src/Clash/Netlist.hs:444:30 in clash-lib-1.9.0-inplace:Clash.Netlist
      mkDeclarations', called at src/Clash/Netlist.hs:403:18 in clash-lib-1.9.0-inplace:Clash.Netlist
      mkDeclarations, called at src/Clash/Netlist.hs:278:40 in clash-lib-1.9.0-inplace:Clash.Netlist
      genComponentT, called at src/Clash/Netlist.hs:241:41 in clash-lib-1.9.0-inplace:Clash.Netlist
      genComponent, called at src/Clash/Netlist.hs:119:9 in clash-lib-1.9.0-inplace:Clash.Netlist
```
which is poor feedback on the fact that I just forgot the second field in last argument of `ConstRepr`.