clash-lang/clash-compiler #2951: Custom bit representation generates wrong Verilog for constant output

kleinreact, 2025-05-03:
The following

```haskell
data T where
  T0 :: Bool -> Bit -> T

{-# ANN module (DataReprAnn $(liftQ [t|T|]) 2
  [ ConstrRepr 'T0 0b11 0b00 [0b10, 0b01]
  ]) #-}

topEntity :: T
topEntity = T0 True low
```

is transformed to  

```verilog
module topEntity
    ( // No inputs
      // Outputs
      output wire [1:0] result
    );

  `include "CheckT0.inc"
  assign result = CheckT0(1'b1, 1'b0);
endmodule
```

with the very efficient, but other than that wrong function `CheckT0`

```verilog
function [1:0] CheckT0;
input v1;
input v2;
begin
  CheckT0 = 2'b00;
end
endfunction
```

christiaanb, 2025-05-03:
Ah, so the issue is more that the compiler doesn’t throw an error that bit-positions are overlapping for a constructor.

kleinreact, 2025-05-03:
Maybe I still don't get the semantics then, but why are they overlapping in this case? The second and third argument to `ConstrRepr` really don't have an influence on the output here, e.g.
```
ConstrRepr 'T0 0b00 0b00 [0b10, 0b01]
ConstrRepr 'T0 0b00 0b11 [0b10, 0b01]
ConstrRepr 'T0 0b11 0b11 [0b10, 0b01]
```
all produce the same.

christiaanb, 2025-05-03:
So you have:
```
ConstrRepr <constructor_name> <position_bits_constructor> <value_bits_constructor> [<position_bits_field>]
```

So if you write e.g.:
```
ConstrRepr 'T0 0b11 0b01 [0b10, 0b01]
```
Then all 2 bits will be used to represent the constructor, in this case the value `0b01`.  Now there’s no more room for the fields, which were encoded to take the MSB for the first field and the LSB for the second field.

christiaanb, 2025-05-03:
If
```
ConstrRepr 'T0 0b00 0b00 [0b10, 0b01]
```
Produces the verilog you posted, that’s a different bug though, as no should be used to represent the constructor.

Perhaps recompilation avoidance doesn’t take custom representations into account, so make sure to use `-fclash-clear -fclash-no-cache` when testing the different variations. If recompilation avoidance doesn’t take custom representations into account, please report a separate issue.

kleinreact, 2025-05-03:
Thanks for the feedback.

```haskell
ConstrRepr 'T0 0b00 0b00 [0b10, 0b01]
```

in combination with `-fclash-clear -fclash-no-cache` indeed produces correct code.

Just that I get the idea of recompilation avoidance correctly. It also should update the file without the `-fclash-clear -fclash-no-cache`, so this is indeed a different bug then?

christiaanb, 2025-05-03:
It should update the file when you change the custom bit representation yeah; without needing those two flags.

kleinreact, 2025-05-05:
Closing this issue in favor of #2953 and #2954.

closed 2025-05-05 by kleinreact