clash-lang/clash-compiler #2963: The `BitSize` of a `Maybe a` with custom representation leads to confusion

pieter-bos, 2025-06-06:
When writing a custom representation for `Maybe Color` of your own `Color` type, the `BitSize` does not match the size of the representation. For example:

```haskell
data Color = R | G | B

{-# ANN module (
  DataReprAnn
    $(liftQ [t|Color|])
    2
    [ ConstrRepr 'R 0b11 0b00 []
    , ConstrRepr 'G 0b11 0b01 []
    , ConstrRepr 'B 0b11 0b10 []
    ]
  ) #-}

{-# ANN module (
  DataReprAnn
    $(liftQ [t|Maybe Color|])
    2
    [ ConstrRepr 'Nothing 0b11 0b11 []
    , ConstrRepr 'Just 0b00 0b00 [0b11]
    ]
  ) #-}

$(deriveBitPack [t|Color|])
```

Gives:

```
ghci> :k! BitSize (Maybe Color)
BitSize (Maybe Color) :: Natural
= CLog 2 2 + 2
```

I've not been able to find a way to get Clash to generate wrong HDL using this fact, I think mostly due to the fact that `pack` and `unpack` are treated as regular functions almost everywhere(^1), so there is no type/size confusion.

Just on the surface it is a bit disappointing I can't derive `BitPack (Maybe Color)` from this representation:

```haskell
$(deriveBitPack [t|Maybe Color|])
-- AppT (ConT GHC.Internal.Maybe.Maybe) (ConT Incoherence.Color) already has a BitPack instance.
```

Furthermore, composite types involving `Maybe Color` generate a `BitPack` that does not compile:

```haskell
data ColorPair = ColorPair (Maybe Color) (Maybe Color)

{-# ANN module (
  DataReprAnn
    $(liftQ [t|ColorPair|])
    4
    [ConstrRepr 'ColorPair 0b0000 0b0000 [0b1100, 0b0011]]
  ) #-}

$(deriveBitPack [t|ColorPair|])
-- • Couldn't match type ‘CLog 2 2’ with ‘0’
--  Expected: BitVector (BitSize (Maybe Color))
--    Actual: BitVector 2
```

Everything does compile if I change the size of the `DataReprAnn` of `ColorPair` to `6`, and give each `Maybe Color` three bits of space. It seems the HDL trashes the extra bit, so the 3-bit representation is not used in that case:

```haskell
topEntity ::
  Signal System (BitVector 6) ->
  Signal System (Maybe Color)
topEntity = fmap $ (\(ColorPair f _) -> f) . unpack
```

becomes:

```verilog
module topEntity
    ( // Inputs
      input wire [5:0] x
      // Outputs
    , output wire [1:0] result
    );

  wire [5:0] c$case_scrut;
  assign c$case_scrut = x;
  assign result = {c$case_scrut[5:3]};

endmodule
```

(^1) the one exception I've encountered is in `deriveBitPack`, which uses a `dontApplyInHDL` to elide its entire `pack`/`unpack` in HDL.

DigitalBrains1, 2025-06-06:
Yeah the fundamental problem is that `BitPack (Maybe a)` already exists. That's why the documentation suggests creating a new `Maybe`-like class that isn't `Maybe`. In that sense the `Maybe Color` example in the same documentation is very flawed, because it suggests doing something that is probably very unwise.

martijnbastiaan, 2025-06-06:
The next generation data repr annotation should probably be generalized such that `BitPack` contains all the functions Clash's backend needs (field projection, constructor selection, etc.).

DigitalBrains1, 2025-06-23:
So my example of `msb (Just R)` is unintuitive but still somewhat law-abiding. True breakage can be had if we make a `BitPack` that deviates from the bit representation of the type itself, and we combine it with `deriveBitPack` (you yourself, Pieter, pointed this out on our company Slack :-). I could easily build something that generates incorrect HDL, mapping R to B and vice versa:

<details><summary><code>UnlawfulBitPack.hs</code></summary>

```haskell
module UnlawfulBitPack where

import Clash.Annotations.BitRepresentation
import Clash.Annotations.BitRepresentation.Deriving
import Clash.Prelude

data Color = R | G | B
  deriving Show

-- Note that the bit representation of 'Color' does not match the 'BitPack'
-- instance
instance BitPack Color where
  type BitSize Color = 2
  pack R = 2
  pack G = 1
  pack B = 0
  unpack 0 = B
  unpack 1 = G
  unpack _ = R

data MaybeColor = JustColor Color
                | NothingColor
{-# ANN module (
  DataReprAnn
    $(liftQ [t|MaybeColor|])
    2
    [ ConstrRepr 'JustColor 0b00 0b00 [0b11]
    , ConstrRepr 'NothingColor 0b11 0b11 []
    ]
  ) #-}

$(deriveBitPack [t|MaybeColor|])

topEntity ::
  Color ->
  Color
topEntity = bitCoerce . JustColor
{-# OPAQUE topEntity #-}
```
</details>

In Haskell simulation, this functions as the identity, but in HDL, it swaps R and B.