clash-lang/clash-compiler #2965: DSL.constructProduct fails without meaningful error message on one-tuple

gergoerdi, 2025-06-08:
I wrote this code:

```
clockToBitF# :: Backend backend => BlackBoxContext -> State backend Doc
clockToBitF# bbCtx
    | [inp] <- map fst (DSL.tInputs bbCtx)
    , [DSL.ety -> resultTy] <- DSL.tResults bbCtx
    = do
        let compName = "ClockToBit"
        instName <- Id.make (compName <> "_inst")
        DSL.declarationReturn bbCtx (compName <> "_block") do
          out <- DSL.declare "bit_o" N.Bit
          DSL.instDecl N.Comp (Id.unsafeMake "ClockToBit") instName [] [("I", inp)] [("O", out)]
          pure [DSL.constructProduct resultTy [out]]

    | otherwise = error $ ppShow bbCtx
```

and it failed without a useful (to me...) error message:

```
<no location info>: error:
    Clash error call:
    Clash.Backend.Verilog(1195): DataCon Bit (DC (Bit,0)) [Identifier (UniqueIdentifier {i_baseName = "bit_o", i_baseNameCaseFold = "bit_o", i_extensionsRev = [], i_idType = Basic, i_hdl = Verilog, i_provenance = []}) Nothing]
    CallStack (from HasCallStack):
      error, called at src/Clash/Backend/Verilog.hs:1195:13 in clash-lib-1.8.1-A6XOGYdXXdQ3s7ZOmGA2Lq:Clash.Backend.Verilog
```

martijnbastiaan, 2025-06-08:
Fair! What are you trying to do? Perhaps we can at least help with a work around.

DigitalBrains1, 2025-06-08:
If I look at the definition of `constructProduct`:

```haskell
-- | Construct a product type given its type and fields
constructProduct :: HWType -> [TExpr] -> TExpr
constructProduct ty els =
  TExpr ty (DataCon ty (DC (ty,0)) (map eex els))
```

And I look at the error, the `TExpr` that `expr_` was trying to evaluate had the "type" `Bit`, since that is the first argument of the printed `DataCon`. In other words, your `resultTy` variable had the value `Bit`. If your primitive produces a one-tuple, it would appear that the fact that this _is_ a one-tuple was already lost somewhere, because it apparently got flattened to just `Bit`? And since `Bit` is not a product type, invoking `constructProduct` on it probably produces a nonsensical result, hence it failing all the defined pattern matches and falling through to a catch-all "You know, I don't know what to do with it, all that's left is to show an error".