clash-lang/clash-compiler #2988: Error parsing blackbox: Clash.Sized.Vector.head

bgamari, 2025-08-09:
The following program fails to compile to Verilog with `clash` `master` (although I suspect that this isn't a new issue):

```
{-# LANGUAGE DataKinds #-}

module Repro where

import Clash.Prelude
import Prelude ()

topEntity
    :: Signal System (Vec 4 (Vec 4 (Unsigned 32)))
    -> Signal System (Vec 4 (Unsigned 32))
topEntity = f

f
    :: Signal System (Vec n (Vec 4 (Unsigned 32)))
    -> Signal System (Vec n (Unsigned 32))
f x = fmap (fmap head) x
{-# OPAQUE f #-}

{-# ANN topEntity
  (Synthesize
    { t_name   = "repro"
    , t_inputs = [ PortName "inp" ]
    , t_output = PortName "outp"
    }) #-}

```

Concretely, the failure is (although this error includes a fair amount of additional context that I have added in https://github.com/bgamari/clash-compiler/tree/wip/blackbox-parse-failure):

```                                                                                       
<no location info>: error:                  
    Clash error call:                  
    error while parsing blackbox: Clash.Sized.Vector.head                               
    in component Hi_topEntity_f                                                         
    error:
    :1:38: error: expected: "[\\", "\\]", "]", "~ACTIVEEDGE", "~AND", "~ARG",
        "~ARGN", "~CMPLE", "~COMPNAME", "~CONST", "~CTXNAME", "~DEPTH", "~DEVNULL",
        "~ENDGENERATE", "~ERROR", "~ERRORO", "~FILE", "~FROMBV", "~GENERATE",
        "~GENSYM", "~INCLUDENAME", "~INDEXTYPE", "~ISACTIVEENABLE", "~ISACTIVEHIGH",
        "~ISINITDEFINED", "~ISLIT", "~ISSCALAR", "~ISSYNC", "~ISUNDEFINED",
        "~ISVAR", "~IW64", "~LENGTH", "~LIT", "~LONGESTPERIOD", "~MAXINDEX",
        "~NAME", "~OTHERSYN", "~OUTPUTUSAGE", "~OUTPUTWIREREG", "~PERIOD",
        "~REPEAT", "~RESULT", "~SEL", "~SIGD", "~SIGDO", "~SIZE", "~STRCMP", "~SYM",
        "~TAG", "~TEMPLATE", "~TOBV", "~TYP", "~TYPEL", "~TYPM", "~TYPMO", "~TYPO",
        "~VAR", "~VARS", "~VIVADO"
    1 | assign ~RESULT = ~FROMBV[~VAR[vec][0][~SIZE[~TYP[0]]-1 -: ~SIZE[~TYPO]]][~>
      |                                      ^                                     
    template:
    assign ~RESULT = ~FROMBV[~VAR[vec][0][~SIZE[~TYP[0]]-1 -: ~SIZE[~TYPO]]][~TYPO];
    
    
    
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/BlackBox/Util.hs:687:5 in clash-lib-1.9.0-inplace:Clash.Netlist.BlackBox.Util

```

Specifically, the problem is the template, which differs from the template in the Verilog:
```patch
+    assign ~RESULT = ~FROMBV[~VAR[vec][0][~SIZE[~TYP[0]]-1 -: ~SIZE[~TYPO]]][~TYPO];
-     assign ~RESULT = ~FROMBV[~VAR[vec][0][\\~SIZE[~TYP[0]]-1 -: ~SIZE[~TYPO]\\]][~TYPO];
```

bgamari, 2025-08-09:
As noted above, the problem here appears to be that something rewrote `[\\` (respectively `\\]`) to `[` (resp. `]`) in the template string. The cause of this appears to be `Clash.Netlist.BlackBox.mkFunInput`, which does some rather dodgy things to construct a black-box from another black-box. Specifically, it uses `prettyBlackBox` to render a blackbox to a string, which it then treats as an identifier. This fails as `prettyBlackBox` fails to escape `[` and `]` correctly.

DigitalBrains1, 2025-08-11:
I think it's the same as https://github.com/clash-lang/clash-compiler/issues/2809 . I had a lot of trouble finding the commit that dealt with this issue, but I was sure I had seen a commit deal with it. Finally thought to search issues instead, and it turns out we never committed the fix that Leon wrote (or started on? Since it never seems to have made the PR stage?). But you can still peek at the code in the [fix-map-head tree](https://github.com/clash-lang/clash-compiler/tree/fix-map-head). I don't know what the state is of the fix.

closed 2025-09-06 by martijnbastiaan

DigitalBrains1, 2025-11-07:
We've released [v1.8.3](https://github.com/clash-lang/clash-compiler/releases/tag/v1.8.3), which includes a fix for this issue.

PR #2989 (2025-09-06): prettyBlackBox: Correctly escape `[` and `]`
The pretty-printer previously failed to round-trip, resulting in invalid primitives due to the logic in `Clash.Netlist.BlackBox.mkFunInput`.

I have also included some clean-ups and error message improvements that I found useful while debugging this.

Fixes #2988.
Fixes #2809.

# TODO
- [x] Add changelog entry
- [x] Add test
commit f2b20fdb (2025-09-06): prettyBlackBox: Correctly escape `[` and `]`

The way that way that blackboxes of function arguments to higher order
blackboxes are handled is somewhat weird.

For historical reasons, they are parsed, pretty-printed, later re-parsed
again. However, the pretty-printer failed to correctly re-escape
previously escaped square brackets, resulting in the generation of
invalid HDL, as seen in #2809.

Fixes #2809.
Fixes #2988.
PR #3001 (2025-09-06): prettyBlackBox: Correctly escape `[` and `]` (copy #2989)
The pretty-printer previously failed to round-trip, resulting in invalid primitives due to the logic in `Clash.Netlist.BlackBox.mkFunInput`.

I have also included some clean-ups and error message improvements that I found useful while debugging this.

Fixes #2988.
Fixes #2809.

# TODO
- [x] Add changelog entry
- [x] Add test<hr>This is an automatic copy of pull request #2989 done by [Mergify](https://mergify.com).
commit 4f5ac601 (2025-09-06): prettyBlackBox: Correctly escape `[` and `]`

The way that way that blackboxes of function arguments to higher order
blackboxes are handled is somewhat weird.

For historical reasons, they are parsed, pretty-printed, later re-parsed
again. However, the pretty-printer failed to correctly re-escape
previously escaped square brackets, resulting in the generation of
invalid HDL, as seen in #2809.

Fixes #2809.
Fixes #2988.

(cherry picked from commit f2b20fdb33cc0bc04e01f8c91fe9330b13671d7b)